cartographer-project/cartographer #433: Serialized SparsePoseGraphs can easily be too big for default proto deserialization

macmason, 2017-07-27:
I'm working with some _very large_ multi-trajectory SLAM problems (10s of kms, indoor), and the resulting serialized `SparsePoseGraph` objects are too big to deserialize, as they're much larger than the default 64MB limit. (Seriously: I'm up to about 1.3GB for the SPG alone.)

There are three solutions:

1. Modify the serialize of SPG to use the pbstream, writing the pieces out one-by-one.
2. Improve pbstream to use a `google::protobuf::io::CodedInputStream` directly, so we can call `SetTotalBytesLimit` with a bigger value.
3. "Mac, you're out of your mind, make yourself a branch and leave the rest of us out of this."

(I've currently implemented 3, via 2, in `macmason/cartographer` and `macmason/cartographer_ros`, in branches both called `mass_map`.)

Thoughts?

FYI @brandon-northcutt

SirVer, 2017-07-28:
Are we talking about [this proto](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/proto/sparse_pose_graph.proto#L32)? Can you figure out what exactly takes up all that space?

Because: having a SPA this big means that sometime before a computer actually had to keep all these constraints (that should be of similar size in memory as in proto) in memory and ran an optimization. This seems like it should take years to optimize - that does not seem too practical.

macmason, 2017-07-28:
Yes, that proto.

The problem is that I have a large number of trajectories, each revisiting the same building, in roughly the same order. As a result, I have _lots_ of loop closures. (The map  looks great!)

For example, my most-recent pbstream serializes 53 trajectories and has 13,206,458 constraints. Optimization takes hours and hours, as expected.

That sounds like a vote for option 3. :-D

wohe, 2017-08-01:
Longer term, option 1 seems the most reasonable to me. In a first step we could write a `Trajectories` proto containing all trajectories as the first message to minimize the diff. We would have multiple messages containing `Constraints`.

Most interesting change would seem to be necessary in `autogenerate_ground_truth_main.cc` which so far is the only place where we are deserializing constraints.