cartographer-project/cartographer #439: Lunar Release?

mikaelarguedas, 2017-08-06:


mikaelarguedas, 2017-08-06:
submitted an empty issue by accident, the content of the issue has now been updated

ojura, 2017-08-06:
Hi! I have never used the packages as it is in heavy development (there was a release for kinetic?), but it should work fine on lunar (I pushed some commits which fix building on 17.04/Lunar).

ojura, 2017-08-06:
Also, there's a related [issue in cartographer_ros (#158)](https://github.com/googlecartographer/cartographer_ros/issues/158).

mikaelarguedas, 2017-08-06:
thanks @ojura for the quick reply.

AFAICT there was a release on indigo but the kinetic release was not possible because ceres-solver was not available for all platforms (related discourse post [here](https://discourse.ros.org/t/ceres-package/872)).

> Also, there's a related issue in cartographer_ros (#158).

Thanks for the pointer, I'll try to comment there as well to get the ball moving

I'm bringing up the request for a release because I had to build ceres + cartographer from source a few times on an ARM board recently and that a binary package would have saved me a couple of hours every time.

I understand the point that an actively developed project would need regular releases for the users to benefit from all the new features. Though from a user perspective I can see a big advantage in being to apt-get install ros-XX-cartographer rather than building from source.

If releasing is of interest to the cartographer developers I'm happy to help if needed

ojura, 2017-08-06:
Disclaimer, I am just a contributor, not a member of the Cartographer team :-) But IIRC they have started making point releases a couple of weeks ago (I think we're now at 0.2.0?)

damonkohler, 2017-08-07:
Also see #204. We're currently doing source releases in the ROS way. We are not pursuing binary releases at this time. @jack-oquin volunteered to do binary releases, but I do not know the status of that effort.

We'll happily accept a [PR to add a Lunar build to Travis](https://github.com/googlecartographer/cartographer_ros/issues/158). At that point, it is defacto supported.

damonkohler, 2017-08-07:
Closing as a duplicate of googlecartographer/cartographer_ros#158

closed 2017-08-07 by damonkohler

mikaelarguedas, 2017-08-07:
Thanks @damonkohler for quick the answer, 
@jack-oquin I looked a bit more into the blockers for releasing and it seems that the packaged version of `ceres-solver` is not packaged properly [for most platforms](https://discourse.ros.org/t/ceres-package/872/18) preventing a binary release for Kinetic and Lunar as long as ceres is not packaged as a ROS package on the buildfarm

ojura, 2017-08-07:
In that case, why not build build ceres-solver as a subproject within libcartographer and ship it together within the binary release?

mikaelarguedas, 2017-08-07:
The main reason I see is that other projects than cartographer within the ROS ecosystem may depend on ceres.
On workaround was to release `ceres-solver` as a ros package and that packages depending on it declare a dependency on `ceres_solver` in their package.xml. But that implies that all packages depending on `ceres-solver` will need to now depend on `ceres_solver`.
Another option is to backport a recent ceres-solver and release it on a PPA (some ROS maintainers started that work and reported progress on [this thread](https://discourse.ros.org/t/ceres-package/872)).
Both these workaround shifts the burden of packaging onto ROS maintainers rather than debian/ubuntu maintainers until we reach the point of having a working upstream `ceres-solver` debian package for all supported platforms.