cartographer-project/cartographer #444: use a tracker per sensor

BrannonKing, 2017-08-09:
Currently Cartographer is limited to a single odometry and single IMU input, even though it takes in a sensor_id with the data. We need separate instances of IMU tracker and odometry tracker for each sensor_id. My two IMU sources produce very different data, and their timestamps are definitely not shared.

Having seen the other discussions on the pose estimator, it may be that it replaces the IMU tracker.

damonkohler, 2017-08-10:
Is this a feature request to support multiple IMUs and odometry sources? That's a pretty complicated prospect for a lot of reasons. Is it not sufficient in your case to just choose the best sensor and use it?

BrannonKing, 2017-08-10:
I'm exploring alternative ways to bring in my wheel encoder data. I have it converted to linear and angular velocities at a high frequency. It's not a far stretch from there to get linear accelerations. When I passed that in as a second IMU, I saw quickly that we were limited to a single IMU instance in trajectory builder.

Two other options that would be helpful:
1. Allow direct input of linear and/or angular velocities.
2. Allow a "relative pose" or "relative odometry" input -- let me put in the x,y,z offset from the last pose output. 

Yes, this is a feature request for one of these three options.

BrannonKing, 2017-08-10:
From the comments on the pose estimator, it sounds like the future is to use the odometry field. Looking into it, it appears that only the odometry differences are used. That should be good enough for now. I'm not going to press this issue.

closed 2017-08-10 by BrannonKing