cartographer-project/cartographer #447: what do I tune for laser shadows? (see pic)

BrannonKing, 2017-08-10:
I have used Cartographer for a while with good odometry and the online correlative scan matcher disabled. When I have good odometry, I never get a picture like the one below. However, I've wanted to utilize Cartographer in situations where I don't have odometry. I have a simple simulation test (with all default settings except for max laser range = 30). My laser runs at 10hz with just a little bit of noise from Gazebo. When I run that using only laser data or laser+IMU, I get pictures like this:

[image]

Does that picture mean that the scan matcher is failing to match the scans? What do I tune to make it more likely to succeed? I've tried playing with the search window parameters, but larger numbers there do nothing but slow down the processing.

BrannonKing, 2017-08-10:
Okay. I understand this now. The tuning guide was helpful, though not complete yet. Please put this info in there (and then we'll close this issue):

If you see shadows or repeated shapes like the above picture, you need to determine first whether they are in the submap or if the lines each represent a different submap. You can do this by disabling the global map: if you see the problem with `optimize_every_n_scans = 0`, then you know the problem is in a single submap. (Make sure that your submap is large enough for the test.) It may also be helpful to output the submap_list ROS topic to verify submap index is never greater than 0.

If the repeated shapes don't exist in a single submap, you need to enlarge the global solver parameters (not covered here).

If the repeated shapes do show in a single submap, it likely has to do with the weights of the solver. You have weighted the odometry or IMU data more than the laser endpoints (aka, occupied space). As mentioned in the configuration guide, you should use the realtime (aka, online) solver only if you have laser data alone (no IMU or odometry) or the default solver (Ceres) fails for you. It's weight parameters are TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.*_cost_weight...

In my particular situation, where I do have IMU data, the Ceres solver is the one I want. Out of the box, it is weighted this way: 

ceres_scan_matcher = {
  occupied_space_weight = 1.,
  translation_weight = 10.,
  rotation_weight = 40., ...

In other words, out of the box we trust the rotational velocities in the IMU 40x more than we trust the laser returns. That doesn't seem quite right for a default setting, but adjusting the weights such that the occupied space (aka, the lines in the above image) to have a more equal footing resolves the repeated shape issue.

damonkohler, 2017-08-11:
@wohe this is another case against #428. Perhaps #428 should be a change to only the backpack configuration.

damonkohler, 2017-08-11:
@BrannonKing thanks for writing up your investigation. I agree that it would be a useful addition to the tuning guide.

BrannonKing, 2017-09-22:
Some of the shadows I've been dealing with were related to the gravity constant messing up pose_estimate's orientation, as I wrote about here: https://github.com/googlecartographer/cartographer/issues/534

I also realized, in reviewing the code, that the weight for the laser scan (er, range data) is applied by a square root function. Hence, if you have 1600 laser scan points and an occupied space weight of 1, you would have to give your translation and rotation each a weight of 40 to give them the same punch as the laser.

ojura, 2017-09-22:
I have just seen this issue. I can also add that #428 completely broke one of my bags (2D rangefinder + IMU), I had to perform regression testing to find out the cause.

cschuet, 2017-10-06:
@BrannonKing: Thanks for the investigation so far. Could provide a bag to reproduce the issue that can be seen in the rviz visualization above? Then I'd be happy to further investigate. My gut feeling is that robot movement is quite abrupt and at high velocity. This might contradict the initial assumed state of the robot standing still so much that the scan matcher cannot convince Cartographer otherwise.

BrannonKing, 2017-10-06:
This is all the same bug, people. The rotation data going into Ceres is complete crap, as I wrote about in #534. Therefore, giving the rotation any value more than featherweight makes for a complete mess. You don't need any particular bag to reproduce the problem. Modify Cartographer to print out the rotation data coming in to the Ceres scan matcher's Match method. It should generally follow the real angle of the vehicle (in geometric radians). It's worse with odometry alone than with IMU+odometry. I'm not sure how well it works with IMU and no odometry, but it will still be incorrect data.

wohe, 2017-10-06:
The assumption that the 2D Ceres scan matcher's input yaw should follow the angle of the vehicle is wrong: It must be understood in relationship to the 'point_cloud' which already has been transformed for gravity alignment which includes an arbitrary yaw component.

Moreover, I'm convinced that the rotation data going into Ceres must be pretty good:
The Ceres scan matcher is finding local optima and the demo bags work very well. This would be impossible if the initial estimate of rotation would be bad.

SirVer, 2017-10-06:
@BrannonKing Brannon, some things do not compute in your analysis. In addition to what @wohe pointed out I see these issues:

> In other words, out of the box we trust the rotational velocities in the IMU 40x more than we trust the laser returns.

This assumption of the weights mentioned above is incorrect - the weights do not have the same units (unfortunately), since the terms are not easy to normalize. Also, the translation and rotation terms costs are for moving away from the initial guess. This guess is not only affected by IMU, but also by prior state.

In your OP you mention that your laser has 10hz and the picture shows 5 duplicated walls quite some distance apart. If those are 5 separate scans your robot starts probably with a substantial speed initially. As @cschuet mentioned this contradicts the initial guess of Cartographer giving the Ceres scan matcher no chance of finding the correct pose. Also you mention the correlative scan matcher, but it is unclear to me when and how you enabled it and what combination of IMU + no IMU show the problem. It would be easier to experiment with this ourselves.

We do not argue that there might be a bug lurking here, but you do not provide evidence as convincing as you seem to think. Please give us a repro case, including configuration and bag (containing IMU + laser data), so we can investigate the failing issue.

BrannonKing, 2017-10-06:
Thanks for giving me feedback today guys. I wrote a possible scenario for reproducing some of my orientation problems over on #534 just barely. I have the bag that I posted on https://github.com/googlecartographer/cartographer_ros/issues/488 some time ago that still has a lot of trouble orienting itself with its odometry and illustrates a lot of problems that I see. We can finish out the conversation on #534 before diving into that other example, though.

SirVer, 2017-10-17:
@BrannonKing I started looking into this, but I can not reproduce your issues with confidence. I downloaded your bag and configuration file from googlecartographer/cartographer_ros#488 and see poor quality when I just run it - as expected. To visualize odometry I wanted to run it online, i.e. not in the offline node - for that I miss a URDF to get the link between `T7_laserscan_plugin` and `base_link`. (Longer story: This information is contained in your bag in `tf_static`, but `static_transform_publisher` seems to have issues with correct timestamping when launched from roslaunch next to rosbag play AND urdf XML does not support quaternions - while rostopic echo only outputs quaternions). Therefore, could you post your URDF so that your example can be run online?

I feel I am still missing the bigger picture. So far I am a bit confused by your setup in this bag - for example your `/truth_odom` publishes `base_link` <-> `T7_truth_plugin`, but you configured Cartographer to expect odometry in the frame `odom` and I see nothing tying these two together. I'd like to run through this online to get a better understanding.

BrannonKing, 2017-10-17:
@SirVer , I recognize that there is a lot of strangeness with the data I posed on 488. It is data captured coming over a ROS2->ROS1 bridge; I never had a URDF for this and there is room for corruption in the data. I've attached a URDF that I think should work, though. And the truth_odom topic should get remapped to odom in the launch file. [bag.urdf.zip](https://github.com/googlecartographer/cartographer/files/1391684/bag.urdf.zip) You may need to duplicate the IMU joint in the urdf file to get the right transform names.

SirVer, 2017-11-07:
@BrannonKing I spend some time now looking into the repo case and bag you posted in https://github.com/googlecartographer/cartographer_ros/issues/488 and this gives me a lot of confidence that there is no issue in odometry handling. There are however significant issues with your data. I am closing this issue for now, but I will have a look at https://github.com/googlecartographer/cartographer/issues/534#issuecomment-334888560 still.

- The collection data in the bag runs at 50% speed, [i.e. plotting `serialization_time` (i.e. rosbage::MessageInstance::get_time()) vs `sensor_time` (i.e. `msg->header.stamp`)](https://github.com/googlecartographer/cartographer/files/1449743/non_realtime_timing.pdf) shows that the slope is != 1. This results in problems in playing this data back, since ROS time runs half as fast as sensor time, confusing TF interpolation and extrapolation.
- Additionally, sensor data timestamps are in 1970, while serialization timestamps are in 2017. Since ros uses serialization timestamp for defining `ros::Time::now` means that you always ask rViz to extrapolate 30 years into the future. This, plus the point above mess up rViz in offline and in online, but do not affect the SLAM result, since Cartographer is purely data driven. This also explains why the map shows up and looks alrightish.
- Though the data is supposed to be 2D, looking at the live playback of the matched point cloud it looks like the robot tilts, i.e. there are ghost walls which yield the laser fans you mentioned in OP. I tried to make [a video of this](https://www.youtube.com/watch?v=QceysCWnYLw). It flickers a lot, but the issue is visible. Unfortunately I cannot look at the laser fan directly, since its data timestamps place it in 1970 which makes transforming the data into a finite domain hard.
- Running offline Cartographer only with laser and later toggling `use_odometry = true` shows that the result gets much improved through odometry. Here I am plotting `x` vs `y`, `t` vs `x` and `t` vs `y` with only laser and with laser + odometry. The data was extracted out of the `.pbstream` and the `/truth_odom` topic.

You can find my changes in https://github.com/googlecartographer/cartographer_ros/compare/master...SirVer:issue/488_534, but they are only: using the URDF, adding an online launch file, modernization of the configuration and hacking in echos to get to the pose results for plotting .

Without odom:
[image]
[image]
[image]

With odom:
[image]
[image]
[image]

closed 2017-11-07 by SirVer

BrannonKing, 2017-11-07:
@SirVer , thank you for your very thorough investigation of this issue. I'm satisfied with your conclusions on this and the other issue concerning non-orthogonal tracking frames.