cartographer-project/cartographer #449: Can't install on ROS Indigo: ros-indigo-tf2-eigen not found

xenotaur, 2017-08-11:
Hey gang, I was attempting to install Cartographer on a Fetch robot running ROS Indigo but the installation failed because of a missing package ros-indigo-tf2-eigen. I've gone through the instructions from scratch twice trying to identify the issue, and my best hypothesis is that perhaps some old repositories are confusing the system by pointing to things no longer available. But I've done my best to clean them out and the system still can't seem to resolve this ros-indigo-tf2-eigen package. 

Thoughts? I am following the instructions at: https://google-cartographer-ros.readthedocs.io/en/latest/
In this iteration, everything works up to the line "rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y"

Example failure:
------------------------------------------------------------------------------------
centaur@fetch8 [Thu Aug 10 17:47:48]: ~/Workspace/Cartographer
$ rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
WARNING: Package name "ceres-solver" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "ceres-solver" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "ceres-solver" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
executing command [sudo -H apt-get install -y ros-indigo-tf2-eigen]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-indigo-tf2-eigen
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-indigo-tf2-eigen] failed

damonkohler, 2017-08-11:
Interesting failure. Do you not see the package listed when you use `apt-cache search tf2`? It sounds like your ROS deb repo isn't properly configured.

damonkohler, 2017-08-16:
As discussed offline, this is an apt configuration issue.

closed 2017-08-16 by damonkohler

SnowCarter, 2018-09-29:
I meet the same issues: can not install ros-indigo-tf2-eigen. Can you tell me how to solve it in details?

vishalsleotechsa, 2019-08-10:
what is the way to solve this issue
ubuntu 14.04 64-bit
16 GB RAM
GCC version 4.8.4
ROS -indigo distribution

I have attached the terminal text,
"
$rosdep install --from-paths src --ignore-src --rosdistro=indigo -y
executing command [sudo -H apt-get install -y ros-indigo-tf2-eigen]
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
ros-indigo-tf2-eigen
0 upgraded, 1 newly installed, 0 to remove and 317 not upgraded.
Need to get 8,234 B of archives.
After this operation, 98.3 kB of additional disk space will be used.
Err http://packages.ros.org/ros/ubuntu/ trusty/main ros-indigo-tf2-eigen amd64 0.5.20-0trusty-20181117-044151-0800
404 Not Found [IP: 140.211.166.134 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-tf2-eigen/ros-indigo-tf2-eigen_0.5.20-0trusty-20181117-044151-0800_amd64.deb 404 Not Found [IP: 140.211.166.134 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
apt: command [sudo -H apt-get install -y ros-indigo-tf2-eigen] failed
"