cartographer-project/cartographer #452: How to run the standalone cartographer?

amiramir99, 2017-08-11:
Hi guys,
Probably, it is a trivial question, but I ask it anyway. I am trying to find an example code to run the 2d depth cartographer (standalone) for turtlebot 2 (kobuki base + kinect RGBD), but I am having trouble finding the main cpp function in the repository. I have already tested the ros version and it is awesome. Your help is appreciated :)
Amir

wohe, 2017-08-16:
Hi Amir, Cartographer itself is a library, i.e. if you are not using ROS, you are expected to write your own `main()` function and call the Cartographer library from your code. The ROS integration should give you an idea how to do that. If you have specific questions, just let me know.

amiramir99, 2017-08-17:
Thank you Wohe! I will definitely have more questions for you! I appreciate that a lot!

closed 2017-08-18 by damonkohler