cartographer-project/cartographer #454: Build and Installation error

krishnachandrajoshi, 2017-08-15:
Hi,

I've been trying to install cartographer on my system and there was an error during the installation because **Project 'laser_loop_closure' tried to find library 'point_cloud_filter'**. Here is the entire error 

CMake Error at /home/ubuntu/catkin_ws/install_isolated/share/point_cloud_filter/cmake/point_cloud_filterConfig.cmake:141 (message):
  Project 'laser_loop_closure' tried to find library 'point_cloud_filter'.
  The library is neither a target nor built/installed properly.  Did you
  compile project 'point_cloud_filter'? Did you find_package() it before the
  subdirectory containing its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:14 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build_isolated/laser_loop_closure/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build_isolated/laser_loop_closure/CMakeFiles/CMakeError.log".
<== Failed to process package 'laser_loop_closure': 
  Command '['/home/ubuntu/catkin_ws/install_isolated/env.sh', 'cmake', '/home/ubuntu/catkin_ws/src/blam/internal/src/laser_loop_closure', '-DCATKIN_DEVEL_PREFIX=/home/ubuntu/catkin_ws/devel_isolated/laser_loop_closure', '-DCMAKE_INSTALL_PREFIX=/home/ubuntu/catkin_ws/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/ubuntu/catkin_ws/build_isolated/laser_loop_closure && /home/ubuntu/catkin_ws/install_isolated/env.sh cmake /home/ubuntu/catkin_ws/src/blam/internal/src/laser_loop_closure -DCATKIN_DEVEL_PREFIX=/home/ubuntu/catkin_ws/devel_isolated/laser_loop_closure -DCMAKE_INSTALL_PREFIX=/home/ubuntu/catkin_ws/install_isolated -G Ninja

Command failed, exiting.

Please help.

Thanks

drigz, 2017-08-16:
This error message seems unrelated to Cartographer - it appears to be created by a project called `laser_loop_closure`, which is not part of Cartographer. I'm not aware of Cartographer depending on `point_cloud_filter`, either.

Have you added anything apart from Cartographer to the workspace that might lead to this?

If you can't resolve the issue, please let us know the exact steps you followed to build Cartographer, including all the commands you ran. Here are the [official instructions](https://google-cartographer-ros.readthedocs.io/en/latest/#building-installation).

krishnachandrajoshi, 2017-08-16:
I already have Velodyne Lidar drivers in the workspace. I don't know how to work around and make them not interfere with each other.

drigz, 2017-08-17:
> I don't know how to work around and make them not interfere with each other.

Is that really the problem? The error doesn't mention interference from Cartographer.

Does the build for the lidar drivers work if you remove cartographer from the workspace?

If it is a conflict, please post a list of everything you've added to the workspace, including links and version numbers, so that I can recreate your environment. So far you have not given enough information to reproduce the error.

SirVer, 2017-10-23:
Closing as this build error is not related to Cartographer.

closed 2017-10-23 by SirVer