cartographer-project/cartographer #459: Lidar sensor suite selection for 3D SLAM

Changliu52, 2017-08-16:
Hi Guys,

I want to use cartographer to perform 3D mapping and 6-DoF state estimation for drone to fly in unstructured environment (no assumption of flat wall or flat floor). Here I have some ideas of choosing the sensors, which I think theoretically could work. Please advise on whether they can work in practice and how ideal they are for the Cartographer to work with in the current stage:

1. Use one velodyne vlp-16 with or without IMU: Is this good for 6DoF recovery?
2. Use one 2D scanning lidar with IMU (the two sensors rigidly fixed together, but together can rotate around): Is this sufficient to recover 6DoF? or Is this sufficient to recover x-y-yaw without flat wall/floor assumption or sensor always horizontal assumption?
3. Use two 2D scanning lidar (one horizontal, one vertical) with or without IMU: Again, no pre-assumptions, hoping it can work like this video: https://www.youtube.com/watch?v=jlkASUo52eM, although this video was testing in flat corridor.

Many thanks in advance.
Chang

BrannonKing, 2017-08-16:
This question is probably better for the news group.

You should definitely include an IMU. Without it you have to use the slower "online" solver. I don't get good performance out of that.

Changliu52, 2017-08-17:
Thanks for the reply. Any link to the news group?
OK. IMU is a must then.

wohe, 2017-08-17:
For 3D SLAM with Cartographer an IMU is a requirement.

Using one VLP-16 might work, but you'd have to make sure you're constrained in all 6-DoF, e.g. by making sure the LIDAR is mounted so that it always sees parts of the floor and some walls.

Changliu52, 2017-08-17:
Hi Wohe, thank you for the answer.
Say if the wall is very unstructured (not flat), then the requirement of including the floor can be eliminated?

So in my 3rd option (Use two 2D scanning lidar, one horizontal, one vertical, with IMU), this setup should give the constraints in all 6-DoF, right?

wohe, 2017-08-18:
Yes, it might work with the right kind of walls. Very small indentations might not be enough though and it should not look the same at different heights like a curtain would.

I'm not sure you can make option 3 work easily since if you move orthogonal to the plane seen by one of the LIDARs, there might not be any overlap with older data in one of the directions. But it is also an easy option to explore because you do not have to build hardware and can just use the 2D backpack data we provide. Feel free to experiment and let me know if you can make it work with 3D SLAM.

closed 2017-08-18 by damonkohler

Changliu52, 2017-08-18:
Interesting! Thank you for the suggestions. I will definitely give it a try. I will update in this thread for any success.