cartographer-project/cartographer #464: Submap updating and maintenance

johnzjq, 2017-08-21:
Hi creators,
I am working on a simple implementation of lifelong mapping. I planned to firstly maintain the spatially overlapped submaps based on their timestamps and to trim the outdated ones, and secondly to insert newly incorporated range data based on the optimized trajectory to the FINISHED submaps for updating.
I would like to know your preferences or future plan for this.

johnzjq, 2017-08-21:
Two problems remain: 1 triming arbitary submap 2 remove old range_data.

closed 2017-08-21 by johnzjq

damonkohler, 2017-08-23:
@johnzjq, could you give us some context for why you closed this? Is there a PR we should look at?

johnzjq, 2017-08-24:
Hi @damonkohler There is none, thanks for your concern=) Initially I opened this issue but gradually found related discussions in some other issues, ie. trimming submap in the middle etc. Therefore I closed this issue for preventing any confusion.