cartographer-project/cartographer #467: decreased map quality after angular velocity estimation(#458)

jihoonl, 2017-08-22:
After merging #458, our map quality significantly decreased even with a large search window of `real_time_correlative_scan_matcher`. It seems like #458 made cartographer relies more on odometry and made harder to use robots with imprecise odometry.
*Before*
[image]

*After*
[image]

I will send you an e-mail to explain how to reproduce it on your side.

ojura, 2017-08-22:
Have you tried readjusting the occupied space, translation and rotation weights in Ceres scan matcher options?

jihoonl, 2017-08-22:
It doesn't seem to help to obtain the same quality map as before.

wohe, 2017-08-22:
How did you determine that #458 is the cause? I'm asking because #456 was merged soon afterwards and also makes use of odometry information. Did you check with #458 but without #456?

jihoonl, 2017-08-22:
Yeah. I double checked that #458 is the cause. It was still breaking without #456.

BrannonKing, 2017-09-15:
Can you retest this with the odometry fixes from last night?

ojura, 2017-09-15:
#533 restored 2D w/ odometry to be pre-#499. This started occuring for @jihoonl even earlier, at #458, so I doubt #533 will help. But it doesn't hurt to try :)

BrannonKing, 2017-09-22:
For the test image above, does that include IMU data or is it only Odometry + LaserScan?

BrannonKing, 2017-09-22:
See my recent comment on the odometric rotation issues here: https://github.com/googlecartographer/cartographer/issues/534

jihoonl, 2017-09-26:
[image]

@BrannonKing Yes, it looks like the latest version mitigated the problem. The quality isn't as good as before though. And it is only odometry + LaserScan.

SirVer, 2017-11-07:
I am closing this as it seems to be mostly a tuning issue after changes to the code. Without data and configuration, this is not actionable.

closed 2017-11-07 by SirVer