cartographer-project/cartographer #469: please add support for an initial world pose

BrannonKing, 2017-08-22:
Please allow us to set the initial value of `pose_estimate_` as discussed here: https://groups.google.com/forum/#!topic/google-cartographer/dqgz9CNCPKA

BrannonKing, 2017-08-23:
I suppose that exposing the quality of the pose_estimate_ may reduce the need for this feature -- I could ignore pose output until I'm sufficiently certain of my position. However, even after that this may be handy for situations where the startup laser scan is ambiguous (imaging facing a tight corner on startup, etc.)

TysonUQ, 2017-09-01:
I too would appreciate the ability to set the initial pose.

I can read in a previously saved map, and eventually (after travelling around for a few minutes) locate myself within it.

It would be great to read in the map and tell Cartographer the pose X_0 (which I know) that the platform should be starting from within it, rather than wait for it to work it out.  Alternatively, would it be possible to apply inv(X_0) to the saved map as it is read in, effectively putting the prior around the (0, 0) starting pose?

Entropy512, 2017-09-11:
I've recently seen some commits referencing the ability to add fixed-frame reference data to the optimization/constraint problem - this might handle the issue of initial pose estimate quality when localizing in an existing map?

cschuet, 2017-09-28:
Right, we are thinking that fixed-frame poses might also the right way to allow people to inject prior knowledge about the position of the system e.g. when it starts up.

I am hoping to get to that next week.

ojura, 2017-09-28:
@cschuet check out my Custom constraints PR - it allows adding constraints which involve nodes and submaps that do not yet exist. It might be what you're looking for.

cschuet, 2017-09-28:
Thanks @ojura I will take a look. In reply to #543 that I just closed to keep the discussion focussed to this issue.

I haven't discussed this with @wohe yet and would like to hear his thoughts but,

* I think fixed frame poses are the right vehicle to allow cartographer clients to inject prior knowledge about where the system is currently located into the problem. Initial pose seems like a special case of that.
* At least for the initial pose use case after having added a fixed frame pose for it, we want local (windowed) searches (at least for a limited time) around that position rather than global searches. If we fail to localize (e.g. because the initial pose was just wrong) we want to be able to recover from that by switching back to global searches.
* The ConnectedComponents and TrajectoryConnectivityState class tracks a well defined property of the optimization problem: are there any / recent global constraints that tie two trajectories together. I would rather not add false information to these to "trick" the system into doing local searches. Rather than implicitly deriving the state from the connectivity state as we do currently I would prefer to model the state explicitly (using some enum) and then add special casing for recent fixed frame poses for initial poses.

Mofef, 2017-09-28:
Thanks for the feedback!
I think initial poses should not be considered in the optimization problem. That's why I would avoid to add constraints.
I looked at the `TrajectoryConnectivityState` as an answer to the question "Do we have good reasons to assume that the relative position of those trajectories is correct?" A recent constraint could be one of those reasons. How about such a state/enum in the `TrajectoryConnectivityState` that tells how the trajectories were connected?

Mofef, 2017-09-28:
Ok, so far it was mostly a gut feeling. Like: constraints should not come from "outside". But on second thought I would argue that we need to assume that the initial pose could be completely wrong. (A human erroneously marked the wrong spot or system woke up on a different spot then it went asleep.) So:
- Compared to e.g. a GPS pose there is no noise model to filter the data (we don't even know the source of the information), and we can't rely on "subsequent initial poses" that would smooth out the error
- A constraint would permanently shift the solution of the optimization problem. Huber loss would reduce the effect, but not eliminate it.
- Deleting the constraint after some time seems hacky to me but might solve the issue, relying on a trimmer (e.g. in localization only mode) is not valid because initial poses might also be used outside of this scenario.

After all, instead of giving a strong connection between two trajectories that should be enforced by the optimization, we rather should give a "hint" about where cartographer should search for such matches in a window of limited time and space. If the hint was bad we should fall back to global search until we receive further hints.

ojura, 2017-09-28:
@Mofef I agree, the issues you raised about lack of filtering constraints inserted this way are valid. But, as you have observed, the optimization problem has this nice property that you can just add or delete a constraint and rerun optimization, and you can get everything fixed. Unlike local SLAM, in which mistakes are baked in forever.

> Like: constraints should not come from "outside".

In principle, I would agree, but I have a use case in which this _does_ make a lot of sense, so that's why I'm pushing for it to get merged :-) It may or may not be a proper solution to the initial pose problem, and I agree about the "hinty" nature that the initial pose functionality should have.

cschuet, 2017-09-28:
Thanks! I would also like to hear what @wohe  has to say about this. He will be back on Monday,

Mofef, 2017-09-28:
Ok I'm also looking forward to his opinion :)

@ojura I would be interested to learn about this use case you have, but lets maybe have this discussion separate. (e.g. in #481 )

About the edit: I meant we would only fall back if we we fail. I.e. if don't find constraints. 
Of course we could also find wrong constraints after getting a wrong hint. That would either be inevitable bad luck (In terms of the two scenarios I described above: The human operator and cartographer were both fooled, or, the robot was kidnapped to a position that looks very similar to the old one). 
It might also be a matter of parameter tuning.

damienrg, 2017-09-29:
From the discussion it looks like there are 2 use cases:
1. trustworthy initial pose (have to be used by cartographer)
2. vague initial pose (can be bypassed by cartographer)
For now, there is fixed frame support only in 3D or with custom constraints via #481 in 2D.
If think that I you implement the initial pose feature you have to support weight coming from the user to support the 2 use cases. If you choose to implement it with fixed frame, as a side effect, it will also permit the user to add fixed frame with custom weight and use fixed weight by option. This is what @ojura does in #481.

apr8, 2017-10-04:
Could this idea #511 also be reviewed? Would love to get feedback on that.

Mofef, 2017-10-13:
@wohe This is the discussion we mentioned in the open house. The two related PRs are #481 and #558 I think you comments would be much appreciated in all those three places :)

PR #606 (2017-11-08): Initial pose implementation.
Fixes #469.
closed 2017-11-08 by cschuet