cartographer-project/cartographer #508: Range data is no longer kept around

ojura, 2017-09-07:
In recent PRs (#503, #488), keeping range data in memory (in trajectory nodes) has been removed from libcartographer. Users of libcartographer who used this data (e.g for realtime map rebuilding as discussed in https://github.com/googlecartographer/cartographer_ros/issues/475, or to implement something like #415) are now out of luck.

Would you consider bringing this back, for example as a configurable option (runtime or compile time)? This way, if you require the range data, you can sacrifice some memory and keep it around; otherwise (and by default), behavior is like in the present implementation, with reduced memory usage. This would be the best of both worlds. Then, for example, the realtime map building functionality would depend on this option being enabled; otherwise, you have to use the assets writer, which retrieves the range data from the bag.

wohe, 2017-09-07:
I think this discussion is already spread out over too many issues. I can just reiterate:

- Memory consumption dropped significantly since we no longer have to keep range data around.
- Real-time rebuilding of submaps for SLAM is not something I see us adding in favor of improved CPU usage and memory consumption; improving submap quality by improving the performance of local SLAM seems like the better approach to me.
- If you are unsatisfied with the current performance of the assets writer, I'd prefer to concentrate on getting this fixed, e.g. as discussed in googlecartographer/cartographer_ros#475.

ojura, 2017-09-07:
Perhaps I focused too much on the map rebuilding thing here -- that is not the only reason why I would like to keep the range data available. I agree that all discussion regarding map rebuilding belongs in other issues.

I wanted to focus in this issue that, as a user of libcartographer, I had been using the range data, and I would like to continue to do so. Of course, I can keep a fork where I retain this. I thought that allowing the library user to decide for themselves whether to keep the range data in memory would be an acceptable compromise which could end up in upstream, thus negating the need for a fork.

wohe, 2017-09-07:
I think I'm missing some context here. Of course you should not have to fork. If you're using range data outside of SLAM why is it not easily possible to keep it around in code separate from the SLAM library? That would seem to achieve better separation of concerns.

ojura, 2017-09-07:
> I think I'm missing some context here.

Sure. For example, among other uses, we are using the point clouds for visualization purposes, and having them stored inside trajectory nodes enabled us to visualize what the rangefinder device had seen at a particular point on the trajectory. 

Cartographer does some processing of point clouds, such as splitting, accumulating and filtering them before they are inserted into a submap. I would like to have access to exactly these, without having to reconstruct them outside of libcartographer by looking them up in the bag and matching them to appropriate trajectory nodes. In other words, exactly what `range_data` in a trajectory node had contained (addressable by its `node_id`).

Perhaps "fork" was too strong a word, but I was trying to avoid the situation where we would have to maintain a modification of libcartographer where we have access to these point clouds. I am trying to find a solution which would enable the library user (us) to retain them, while being minimally invasive to your codebase (as usual). 

The first thing that occurred to me was in the suggestion above, which would reintroduce saving the point clouds in trajectory nodes just like before, controllable by the library user (e.g. `range_data` in `constant_data` could be a null pointer if saving is disabled). Another idea would be to delegate saving the points clouds to the library user, for example, by allowing them to pass a callback function to libcartographer. Before `gravity_aligned_range_data` in `AddAccumulatedRangeData` goes out of scope and is destroyed, the callback function is called and the library user can keep the range data for themselves (for example, by handing a shared pointer to the user, along with the node id). 

The first approach seems better to me, but either way I'll be happy as long as I don't have to fork :-)

wohe, 2017-09-08:
I agree with you. Understanding how exactly the point clouds that are added to the pose graph were seen by Cartographer should be possible.

We currently have a function `mapping::TrajectoryBuilder::pose_estimate()` for that purpose, but maybe it has to be improved. It gives access not only to the range data that made it to the pose graph, but also gives access to range data that was used in local SLAM but did not pass the motion filter. Maybe it should directly expose this information, i.e. a boolean whether this data was filtered out or not. What do you think?

ojura, 2017-09-11:
I have overseen that the last processed range data is saved in `pose_estimate_`, which is exposed by `mapping::TrajectoryBuilder::pose_estimate()`, as you have pointed out. However, suppose that you want to systematically keep _every_ point cloud for later use and the associated trajectory node ID (just like we had before). This is because we want to be able to visualize the past rangefinder data as well.

I do not think that repeatedly polling `pose_estimate()` is appropriate for achieving this, because you do not have direct control over dispatching of the ordered sensor data queue. E.g. after you send rangefinder data to SLAM, it may not be immediately processed. Then, if you call `pose_estimate()`, you would get old data. Then, suppose that multiple rangefinder scans are processed in one go - you might not have the chance to call `pose_estimate()` inbetween in order to retain the resulting range data. Thus, I do not believe that `pose_estimate()` is a viable solution for my concern.

I do not see a third solution - either we enable optional retainment of rangefinder data inside libcartographer (a solution that I am more in favor of, because it means less work for me :-) ), or add a possibility to pass a callback in which we will save the data at the exact moment it is available (and that would be somewhere around when `pose_estimate_` is updated).

ojura, 2017-10-05:
@SirVer in googlecartographer/cartographer_ros#475, you have suggested to open a separate issue for real time map building. I think this issue is relevant. 

To summarize - for us, keeping full range data has two important uses:
- the ability to regenerate a high-quality map in real time (the functionality that has been removed in googlecartographer/cartographer_ros#436 - I have been maintaining it in on top of cartographer_ros head in [a cartographer_ros fork](https://github.com/larics/cartographer_ros/tree/restore-assets-writer))
- the ability to visualize what the robot had seen from any part of the trajectory, and exactly what was inserted in the occupancy grid.

As a temporary solution, I have been maintaining [a libcartographer fork](https://github.com/larics/cartographer/tree/restore-range-data) in which I still keep the original point cloud in each node.

I have proposed two solutions:
  - restore old behaviour (keeping full unfiltered range data in nodes), but make it optional (e.g. have a null pointer if retaining range data is disabled), so you get reduced memory consumption benefits if you don't need this
- make it easier for libcartographer users to retain the range data for themselves, for example by allowing them to pass a callback to libcartographer. Before the original range data is discarded, the libcartographer user gets a chance to retain the range data.

I would prefer option one more, because I believe it's less complex. Just make it a pointer which can be null and add an option to e.g. trajectory builder.

ojura, 2017-10-05:
I also have one additional idea - perhaps higher memory use could be mitigated by immediately dumping the range data to hard drive instead of keeping it in memory. I see that as a possible enhancement of option #1 above.

SirVer, 2017-10-05:
After a quick discussion with @cschuet and @wohe we think option 2. For one it actually enables option 1 trivially, it would also fully support the current use case in cartographer ROS (i.e. publishing point clouds of inserted data) and is therefore just a refactoring away there. Your last comment actually is a likely use case and is also easily expressible through option 2.

We believe this would be a callback function that takes the scan and a bit that says if it was inserted into the pose graph or not. @ojura Do you prepare the small PR enabling this?

ojura, 2017-10-05:
Okay. The callback will be an `std::function<sensor::RangeData, bool inserted>`?

SirVer, 2017-10-05:
sounds about right. Avoiding the [bool in a public API](https://ariya.io/2011/08/hall-of-api-shame-boolean-trap) the second parameter should probably be a `public enum class WasInserted { kYes, kNo };`

ojura, 2017-10-05:
Thanks on the link, that was an interesting read.

Oh, and welcome back :-)

SirVer, 2017-10-05:
> Oh, and welcome back :-)

Thanks :). Glad to be back

ojura, 2017-10-05:
Here are some some additional thoughts. Do we really want the callback to be executed for non-inserted scans? It complicates things a bit. 

One thing I absolutely want to maintain is the association of the range data with trajectory nodes (previously, this was trivial because it was stored inside them). Now, this means that the callback will have to be executed from `SparsePoseGraph::AddScan`, because that's the point when the inserted scan is first assigned a `NodeId`. We can pass the inserted range data in the insertion result to `AddScan`, execute the callback from there when we know the `NodeId` and then discard the scan. If we did not insert the scan, then we would have to execute the callback from the local trajectory builder, because in that case a null pointer is returned for the insertion result. If we did not care at all about non-inserted scans, this would make our life easier.

SirVer, 2017-10-06:
> Here are some some additional thoughts. Do we really want the callback to be executed for non-inserted scans? It complicates things a bit.

@wohe Do you have an opinion on that?

ojura, 2017-10-06:
Actually, it might be more elegant to do this from the global trajectory builder (it owns the callback), and to have `SparsePoseGraph::AddScan` return a `NodeId`.

ojura, 2017-10-06:
#574 implements this for inserted scans only, PTAL.

JJWMr, 2017-10-23:
Hi,guys! I'm working on a single library of cartographer without cartographer_ros and I get probability grid map from all the range_data. However, the range_data is no longer kept around now. How can I get probability grid or occupancy grid from somewhere else? @wohe @ojura

SirVer, 2017-10-23:
@JJWMr You need to use the soon-to-be-merged callback implemented in #574.

JJWMr, 2017-10-23:
Thanks a lot. I'll check it. @SirVer

ojura, 2017-10-23:
It still needs some work on it though, I expect to finish it tomorrow.

cschuet, 2017-10-23:
@wohe will be back in the office on Friday. Given that he has already started the review with you, @ojura, it makes sense to let him conclude it. So probably Friday rather than tomorrow.

ojura, 2017-10-23:
Of course. This was more of an estimate when I would be able to prepare it for a new round of reviewing, not when it would get merged :-)

SirVer, 2017-11-15:
Fixed through #574.

closed 2017-11-15 by SirVer