cartographer-project/cartographer #520: how to save a correct map in cartographer ??

yinnxinn, 2017-09-12:
when i run the demo_revo_lds.launch using official data . everything's OK . 
         i save the map use "rosrun map_server map_saver -f map" ...got the map without occ  ... ..as follows : 
[image]

         and when i use the rosservice call /finishTrajectory "stem:'map' " saved the map . I got the error as follows :+1: 
     **# ERROR: Unable to load type [cartographer_ros_msgs/WriteState].
     Have you typed 'make' in [cartographer_ros_msgs]?**

   it's the version problem resulted in this problem ?????
   shoud I rewrite the map_server source code to save an available map ????
   
   i would appreciate your answers .. thanks ~~

ojura, 2017-09-12:
There is another issue about this, see
https://github.com/googlecartographer/cartographer_ros/issues/475#issuecomment-321261237

To summarize: either patch your map_saver as described in the link above, or use the assets writer.

cschuet, 2017-09-28:
Closing. Thanks @ojura.

closed 2017-09-28 by cschuet