cartographer-project/cartographer #524: Does cartographer work with RGB-D camera?How?

ghost069, 2017-09-13:
I tried the demo and it came out successfully.
[image]
However I want to apply cartographer with RGB-D camera ,but it's all about laser data.I thought it's not going to work with RGB-D camera but the documentation never specifically point out it.(Or maybe it's just me...)
So,anyway,if I want to construct a map like the one below,is it possible with cartographer?If it's possible,what are the extra work that i should do to construct a map like this?I mean,with color,shape and all the other stuff.
Thanks a lot for your kind help!
[image]

Entropy512, 2017-09-14:
It does support depth cameras - I'm not sure if it uses the color information at all.  If you follow the commit history, you'll see that quite a few of the inputs to Cartographer were renamed to be more generic since not all sensors that generate point clouds are lasers.

https://github.com/googlecartographer/cartographer_turtlebot

cschuet, 2017-09-28:
It does support depth cameras in theory, but the results are quite bad. We believe this is due to RGB-D cameras having a different noise model than laser scanners. I have opened #545 to track this issue.

closed 2017-09-28 by cschuet