cartographer-project/cartographer #525: memory leak possibility in cartographer

jihoonl, 2017-09-13:
[image]

[image]

It seems like memory consumption accumulates over time. I let cartographer run in pure localization mode more than 10hrs and the memory consumption permanently increased. Potentially, memory leak somewhere?

ojura, 2017-09-13:
I am suspecting this as well. I often need to restart rviz.

wohe, 2017-09-13:
1. The RViz plugin is unrelated to this. If you restart RViz and that fixes your issue, this is pretty strong evidence that it is unrelated to memory consumption by the Cartographer node. @ojura If you think there is an issue with RViz as well, could you file a separate issue? I would suggest to do some debugging first though to verify that `cartographer_ros` is the likely cause, not e.g. graphics driver issues.
2. Before we jump to any conclusions, we should get more data. @jihoonl Could you try to get something like http://goog-perftools.sourceforge.net/doc/heap_profiler.html to work? It would be very useful to know what is consuming memory.
3. Note that growing memory consumption is expected. While we have removed the largest consumers of memory, we still keep the odometry forever for example. Even in the absence of odometry I expect memory consumption to accumulate, although I would guess a lot slower: `mapping::NestedVectorsById<>` keeps some data, member functions of `SparsePoseGraph` return nested vectors.

ojura, 2017-09-13:
Whoops, I completely missed that it was the cartographer_node in `top`. Sorry. I will open an issue in the other repo. (By the way, if you haven't tried it, I highly recommend `htop` instead of `top`. You might need to install it using the package manager.)

jihoonl, 2017-09-13:
heap profiling sounds good idea. I will try it. And note that I was not using odometry in this case. I agree that it could have been faster if odometry is in use.

cschuet, 2017-09-28:
Closing. All further discussions on [#505](https://github.com/googlecartographer/cartographer_ros/issues/505). Thanks!

jihoonl, 2017-09-28:
@cschuet #505 is a different symptom and cause. Could you reopen this?

As @wohe mentioned, this issue has nothing to do with rviz. It is an issue for memory leak in cartographer.

cschuet, 2017-09-28:
Sorry I missed that. Did you have a chance to do heap profiling?

jihoonl, 2017-09-28:
I haven't gotten a chance to investigate this issue yet due to the other prioritized works. But, I will update the issue as soon as I find something.

cschuet, 2017-09-28:
Cool, thank you.

jihoonl, 2017-12-06:
[image]

Small updates on this issue.  We ran the latest version of cartographer and confirmed that memory consumption still increases in pure localization mode.

```
...
  - 
    trajectory_id: 0
    submap_index: 6
    submap_version: 188
    pose: 
      position: 
        x: 5.22731129042
        y: 3.52407170286
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.00519836008455
        w: 0.999986488435
  - 
    trajectory_id: 3
    submap_index: 52
    submap_version: 40
    pose: 
      position: 
        x: -0.145805460213
        y: 0.465069679046
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.999989935313
        w: 0.00448656576561
  - 
    trajectory_id: 3
    submap_index: 53
    submap_version: 22
    pose: 
      position: 
        x: -0.0236020480483
        y: 0.370183922529
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.999993183655
        w: 0.00369224092749
  - 
    trajectory_id: 3
    submap_index: 54
    submap_version: 2
    pose: 
      position: 
        x: 0.620850280505
        y: -0.111249160334
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.999989809537
        w: 0.00451451240516
---
```

cschuet, 2017-12-06:
@jihoonl Thanks for bringing this up again. At what rate did you see memory increase?

jihoonl, 2017-12-06:
The memory consumption increased from less than 5% to 49%  after about 40 hours of operation. Note that robot was standing most of time.

gaschler, 2017-12-07:
Just a small update, I did a quick check with AddressSanitizer using the script in #745 if it can detect memory leaks or issues in the integration tests, but there was no result. Of course, this would only find actual leaks, not building up larger data structures over time.

gaschler, 2017-12-08:
In order to document the discussion from yesterday, you could track this down by

1. serializing the state after 5 minutes and, say, 2 hours and compare the states.
2. running the heap profiler suggested above by @wohe.
3. send us a reproducible example.

I will write an integration test to do pure localization with a little fake data, but it will (for now) only cover the cartographer side.

I assume that the global SLAM work queue can keep up and submap data is not requested by rviz or the like in your setup.

gaschler, 2018-03-29:
Closing for inactivity, please re-open when there is something new.

closed 2018-03-29 by gaschler