cartographer-project/cartographer #527: num_laser_scans = 2 breaks real time scan matching?

Mofef, 2017-09-13:
I'm not at all sure if this is a problem with our sensor data or with cartographer. 
Out of interest we wanted to try the feature of feeding two different laser scan sources, even though we have an aggregated scan topic containing the scans of both lasers. Surprisingly the resulting map was mostly devestated 
[image].
When `use_odometry = true` (and a version prior to #499) this does not happen, neither when we use the aggregated laser scans (without odometry). Also when using just one of either laser topic the map looks reasonable.

Is there a known case where 2 laser scan topics were used in 2D mapping (without odometry and imu)?

wohe, 2017-09-13:
It would help us tremendously if you provide the following so that we can easily
reproduce your issue:

1. A link to a Git repository containing a branch of `cartographer_ros`
   containing all the configuration, launch, and URDF files required to
   reproduce your issue.
2. A link to a bag file we can use to reproduce your issue.

We will likely be unable to help you without both.

ojura, 2017-09-13:
I have also opened an issue in #515 about #499 causing problems.

Mofef, 2017-09-13:
Ok i'll prepare that

@ojura, yes we also have this issue as @jihoonl mentioned. But its not directly related to this one. It's just that odometry solves this issue when its not broken due to other problems (like PR 499 in our case).

Mofef, 2017-09-13:
@wohe I sent you a mail with the data.

wohe, 2017-09-14:
Thanks. I investigated and it seems a configuration issue which can be resolved by setting:
```
TRAJECTORY_BUILDER_2D.scans_per_accumulation = 2
```
Otherwise local SLAM is scan matching data from the two lasers separately. Even better: subdivide and accumulate, e.g. setting
```
  num_subdivisions_per_laser_scan = 10,
```
and
```
TRAJECTORY_BUILDER_2D.scans_per_accumulation = 20
```
Please re-open if you still have issues getting it to work.

closed 2017-09-14 by wohe

Mofef, 2017-09-14:
Awesome, Thanks! Actually the subdivision and accumulation was exactly the reason why I wanted the scans for each sensor to be separate. But I didn't know that I need to adjust TRAJECTORY_BUILDER_2D.scans_per_accumulation to the number of range finders

AndroidOpencv, 2023-11-24:
IT does not work with me ,I did not use odom .imu and pose extrapolator ,here is my lua file https://github.com/AndroidOpencv/shiyanlou001.git  the bag is 链接link：https://pan.baidu.com/s/1QmZla2u-7Gn8S1Znmr_OzQ?pwd=xwbi 
提取码 password：xwbi  . when the robot go forward ,the scan_matched_points2 did not match ,the map is wrong. when the robot rotate in place,it seems like to slam map better.  I already set num_laser_scans = 2 TRAJECTORY_BUILDER_2D.scans_per_accumulation = 10 TRAJECTORY_BUILDER_2D.scans_per_accumulation = 20  
thank you so much 13210849552@163.com  CHINA 微信 86730825

AndroidOpencv, 2023-11-24:
> Thanks. I investigated and it seems a configuration issue which can be resolved by setting:
>
> ```
> TRAJECTORY_BUILDER_2D.scans_per_accumulation = 2
> ```
>
> Otherwise local SLAM is scan matching data from [...]
 I already set num_laser_scans = 2 TRAJECTORY_BUILDER_2D.scans_per_accumulation = 10 TRAJECTORY_BUILDER_2D.scans_per_accumulation = 20  scan_1 scan_2 laser_1 laser_2 ,it does not work