cartographer-project/cartographer #528: Alignment issues with Eigen

damienrg, 2017-09-13:
With gcc 7.2 I get a segfault when I launch cartographer. After forcing a debug build, I see that it is an alignment problem.
[Fixed-size vectorizable](http://eigen.tuxfamily.org/dox-devel/group__TopicFixedSizeVectorizable.html) object in Eigen must be aligned correctly but this is not enforced in cartographer.
For e.g:
`ActiveSubmaps` contains:
`std::vector<std::shared_ptr<Submap>> submaps_;`
As a `Submap` contains a `Rigid3` which contains a `Quaternion` there is a problem as stated in [Explanation of the assertion on unaligned arrays](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html).

There are multiples fix available but I vote for defining alias of STL containers with Eigen::allocator.
What do you think?

damienrg, 2017-09-14:
Just to add that in the case described in the previous message, making an an alias of std::vector would not fix the problem as the problem is the lack of a [aligned new operator](http://eigen.tuxfamily.org/dox-devel/group__TopicStructHavingEigenMembers.html) in Submap.

wohe, 2017-09-14:
Please see #34 and similar issues for the discussion first.

damienrg, 2017-09-14:
Sorry I did not see #34. I am on x64 so I need to investigate further.

closed 2017-09-14 by damienrg

damienrg, 2017-09-14:
Well the problem was due to that Eigen 3.3 wants to have things aligned on 32 bytes by default when AVX is enabled.

wohe, 2017-09-18:
Interesting. I would not expect a significant performance benefit from AVX, but I have not measured it. Unless you see the need for it, I'd disable it for now. Obviously, something like http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1409 would be a better solution.

ojura, 2019-01-02:
Is this the definitive answer - Cartographer will not support AVX alignment, ever?

Even if we don't need AVX in Cartographer (although I guess it couldn't hurt), suppose that the client code wants to use AVX with Eigen for some other stuff, therefore needs to compiled with e.g. `-march=native` or `-march=corei7-avx` or whatever. The moment we pull in Cartographer's headers, it can be game over because of alignment issues.

libcartographer users would be forced to separate interfacing with Cartographer into a separate library which has AVX disabled privately, and which would include Cartographer headers privately, while the rest of the codebase could use AVX. Perhaps this could work, but I can see it causing quite a bit of pain.