cartographer-project/cartographer #534: Odometry in optimisation decreases quality, makes last submap pose jumpy

ojura, 2017-09-15:
Now that the big bug from #515 is fixed, I would like to point out that I have also observed a change in quality after #456 by @Mofef  ([before](https://user-images.githubusercontent.com/9062353/30225110-934105ea-94d1-11e7-89a6-3a0ce882b107.png), [after](https://user-images.githubusercontent.com/9062353/30225117-97517480-94d1-11e7-84cd-51da5b866c22.png)). Notice that the drift accumulated in the big loop has increased, and it is larger than the window for global scan matching, therefore the loop is not closed. Commenting out `parse_pose_graph_->AddOdometerData()` in `GlobalTrajectoryBuilder::AddSensorData(const sensor::OdometryData&)` yields previous results.

Besides the increased drift in the big loop (which was previously successfully closed), during slamming, you can observe hops in the orientation of the last submap pose, like in [this video](https://github.com/googlecartographer/cartographer/files/1306491/jump.mkv.zip). For comparison, [here is a video](https://github.com/googlecartographer/cartographer/files/1306517/without_jump.mkv.zip) in which odometry is not used in optimisation.

The dataset is from #453, with [this](https://gist.github.com/ojura/e1bfeef9216e9bec0ba8abcfa24ebe4f) configuration file.

Perhaps we need some additional parameters to adjust how much emphasis optimisation puts on odometry (or to disable it altogether, if it produces better results)?

wohe, 2017-09-15:
Did you retune `consecutive_scan_*_penalty_factor`? For example, if the intra submap constraint residuals are generally large, these values would be too large and should be decreased.

Mofef, 2017-09-15:
That would mean that (assuming the code is bug-free) each trajectory node is closer to the last node than to the last node+odometry. That would seem weird to me. But I agree that you might have to re tune some parameters a bit it was the same for our robots.

btw: I saw the commented out lines about the real-time scan matcher. We made very good experience with using it together with odometry and a very samll search window. As a rule of thumb: Set the search windows to something like the amount of "worst case odometry drift" you expect from driving a distance of (max_speed / laser_frequency) meters +10..50%

ojura, 2017-09-18:
Thanks on the suggestions. Actually, for the use case at the moment, it's actually _good_ that I have some drift, so fixing this is not a high priority. I will update the issue later and report if I was able to restore the performance on this bag to pre-#515.

BrannonKing, 2017-09-22:
I've made some progress tracking this issue down today. It started with outputing the `pose_estimate` that goes into `ceres_scan_matcher::Match`. In my testing with only Odometry + LaserScan, the x,y estimate is good and the yaw estimate is always close to zero. There are at least three problems involved. We'll take them one at a time:

## Problem one: the death of the pose_estimate's rotation vector.
This problem occurs in `mapping_2d/LocalTrajectoryBuilder::AddAccumulatedRangeData`. Inside this method, we multiply our fresh pose estimate by the inverse of our gravitational alignment vector like this: 
```cpp
  const transform::Rigid2d pose_prediction = transform::Project2D(
      non_gravity_aligned_pose_prediction * gravity_alignment.inverse());
```
Unfortunately, it's not really a gravity alignment vector. If you track down where it came from (in `PoseExtrapolator::EstimateGravityOrientation`), you will see that it was actually just a copy of our most recent IMU orientation accumulation. That accumulation, if everything is working correctly, is pretty darn close to our estimated orientation. It's understandable, then, that multiplying by its inverse demolishes our pose_estimate's rotation whilst leaving x,y untouched.

## Problem two: PoseExtrapolator::ExtrapolateRotation.
Observations:
1. This method is intended to return the rotational difference from now to the last "timed pose".
2. If you have a decent IMU, you will only get the difference to the last IMU update, which may be very minute.
3. If you don't have an IMU, you will always get an identity quaternion out, because that data only changes when we get new "timed poses". 
4. There is some risk that the compiler would optimize out the memcpy inside that method, thus always giving you a identity quaternion even when you have an IMU.

The way the code is at the moment, we substitute inputs into the IMU tracker when it doesn't have any real data. We build this data from computed pose estimates. I think that we should not do that; that complexity is obscuring this problem. Keep the IMU tracker for IMU data only. Then, if the tracker is empty when we need rotation info, substitute the computation using the odometry estimates.

ExtrapolateRotation and its partner should be named more aptly and should probably take the original pose as a parameter rather than relying on class state (and doing the `back()` call twice).

I couldn't see that we had any unit tests for PoseExtrapolator. That's pretty scary. We should have one where we rotate a quarter pi per second using odometry only and get all eight pie pieces out.

## Problem three: RotationDeltaCostFunctor is not continuous.
I believe that a simple difference of angles is insufficient. You will have hops in the data if the angles wrap. I think that we need to find the smallest difference. For that, we need code like this:
```cpp
  bool operator()(const T* const pose, T* residual) const {
    T a = pose[2] - angle_ + M_PI;
    auto n = M_PI * 2.0;
    T mod = a - ceres::floor(a / n) * n; // ceres version of fmod
    residual[0] = scaling_factor_ * (mod - M_PI);
    return true;
  }
```

ojura, 2017-09-23:
The first section's title reads as if it were from some kind of a SLAM novel :-) Regarding the gravity_alignment, have you seen the discussion from #515 where I was also a bit confused by its role?

https://github.com/googlecartographer/cartographer/issues/515#issuecomment-329158429

BrannonKing, 2017-09-23:
@ojura , I had not read all your comments there. Thanks for pointing that out. It appears that you did understand that using an identity transform in place of the gravity estimation vector improved results significantly. However, that issue was closed without sufficiently resolving that concern: we still get much better results with the identity transform, and `EstimateGravityOrientation` still appears to do nothing related to gravity. Actually, it appears that IMUTracker is utilizing its own gravity estimation inside. Therefore, we shouldn't need to apply any gravity transform in `AddAccumulatedRangeData` because it already came through in the orientation estimation.

BrannonKing, 2017-09-25:
So my open question is:
Should `PoseExtrapolator::EstimateGravityOrientation` return `ImuTracker::gravity_vector_` or should `PoseExtrapolator::EstimateGravityOrientation` go away, thus relying on ImuTracker to encapsulate the gravity manipulations?

BrannonKing, 2017-10-02:
@wohe , 

I'm still anxious for some feedback on the right way to repair the gravity-orientation-stuff-kills-the-rotation-estimation-stuff bug. This issue affects us severely. The current Cartographer codebase is unusable with this bug; we run a build from a year ago because of this. I recognize that we've had some fun robotics conferences lately and that Oktoberfest is in progress. Please, though, keep this at the top of the priority list!

ojura, 2017-10-02:
How come you're using such an old build? `EstimateGravityOrientation()` got merged about a month ago (#499).

BrannonKing, 2017-10-02:
Odometry has been broken for ages. It was broken before the gravity thing broke it worse.

ojura, 2017-10-02:
I see. Still, if you can pinpoint the exact PR where odometry broke apart for you, it would be probably helpful. (I will use this oportunity to plug [this repo](https://github.com/larics/cartographer_combined) that I maintain, which contains both `cartographer` and `cartographer_ros` in a single chronological commit tree, so you can easily do a git bisect and find the regression.)

wohe, 2017-10-06:
Please note that our demos at https://google-cartographer-ros-for-fetch-robotics-platforms.readthedocs.io/en/latest/?badge=latest use odometry, so I'm convinced that any bug that might exist cannot make it "completely broken".

Regarding your problem one: This is a misunderstanding of what `gravity_alignment` does. It is a rotation that takes data from the tracking frame to a frame in which +z is the direction of gravity. This fixes two degrees of freedom. The third degree of freedom is unspecified and you should not make any assumption about it. That it might have a yaw component similar to  `non_gravity_aligned_pose_prediction` is possible but irrelevant. `non_gravity_aligned_pose_prediction` contains roll and pitch, so we must remove this part before projecting to 2D without introducing a large error. The 2D pose prediction is the transform from the gravity aligned frame to the local map frame.

BrannonKing, 2017-10-06:
Thanks for further explaining the purpose of the gravity_alignment vector. My assumptions about it were incorrect. I did some more testing today using your "fetch" sample data and the latest code. I could not break that sample as long as I included its IMU data. I still contend that using odometry without IMU is broken. However, you don't see the breakage without a contrary orientation in the tracking frame. Try this using the sample "fetch" data with these modifications: 
1. Modify the freight.lua file to use `rotated_imu` for the tracking frame and to have IMU data disabled (`use_imu_data = false`).
2. Modify the offline_freight_2d.launch file to include `-urdf_filename /home/user/ros1_ws/tester.urdf`.
3. Create this tester.urdf file:
```xml
<robot name="freight">
  <link name="base_link" />
  <link name="rotated_imu"/>
  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="rotated_imu" />
    <origin xyz="0 0 0" rpy="0 1 1" />
  </joint>
</robot>
```
Do you expect that scenario to work? It doesn't at the moment. The "fetch" data is very tame from a transform perspective. I don't have any sample data where the IMU and odometry and laser are collocated. No two are in the same spot. My IMUs are never aligned with the vehicle axis, either (thanks to space constraints), and my lasers are always upside down.

SirVer, 2017-11-07:
@BrannonKing No, I would not expect this to work. In 2D without IMU the tracking frame has to be z-aligned, orthogonal to the plane of movement.

Rational: If we have an IMU, we can use gravity to project our laser points into the plane of movement. But if you take the IMU away, we have to trust your URDF to bring the laser points into the plane of movement. 

The extreme would be if you gave Cartographer a `tracking_frame` with a z-axis parallel to the plane of movement: All laser fans would then end up on a single line after transformation.

So your example cannot work. I am not sure what this has to do with odometry though.

I  reread https://github.com/googlecartographer/cartographer/issues/534#issuecomment-331566003 and quickly want to rebuddle:

Problem one was addressed by @wohe above. Problem two does seem all correct to me - it just means that if you have a sub-optimal odometry vs imu packet distribution you might extrapolate farther than you'll like. Not a bug though. Problem 3 is not a problem since this cost functor is used to track a delta from a prediction, i.e. will always be around zero where it is perfectly continuous.

I feel I did a thorough investigation of the issue discussed in here. I am closing this issues, since I found nothing actionable. 

@ojura Your OP was voicing concerns about odometry not bringing improvements in all cases. I think for your particular issue this is mostly about tuning of new features. Since it has been a I guess you found better tuning options. If you still think more options for tuning odometry or discussion around tuning is required, could you open a new issue?

closed 2017-11-07 by SirVer

ojura, 2017-11-07:
I still haven't tried tuning `consecutive_scan_*_penalty_factor` as @wohe suggested. I will update when I do.