cartographer-project/cartographer #535: Still suffer from slightly drifting using KITTI data with RTK+RT3000

johnzjq, 2017-09-15:
Hi, I am trying to use cartographer building the outdoor map with KITTI raw data. Since the scanmatch 
  in LocalTrajectoryBuilder constantly produces erroneous pose estimation in street scenarios, I just bypass the scanmatch here, and hope the high precision pose data produced from RTK+RT3000 would still help guarantee the quality of the map. However, when the car came back to a loop, the street will drift randomly at around 5 meters...
I am using the odometry data and fixed pose data from KITTI bag, is there any clue about the causes?

BrannonKing, 2017-09-18:
Have you tried it without odometry? Using the real-time scan matcher rather than Ceres?

johnzjq, 2017-09-21:
Thanks! Thinks has improved much since #533 ,  before that fix I have to skip the spg add_odometry(). I 've tried the real-time scan matcher. Without odometry it cannot produce acceptable results. With odometry data, the performance of real-time version is still inferior to that of ceres version. However, the results show that odometry data, though captured from high precision INS system, still drifts (of around 3-5 meters) and require scan matcher to correct the predicted poses (most of the time it can). I am afraid whether the odometry data was not corrected by the rtk? or the calculated time slice for dead reckoning is not accurate enough? I will investigate ways to use fixed frame poses directly.

SirVer, 2017-10-23:
The KITTI data is hard for us since we do not fully support pure scan matching and they threw away most of their IMU data - sampling it at only 10 hz, instead of the 100 Hz the IMU was providing. You can get decent maps out of it, but they are not super great.

See my comments here. I also posted my tunings there:
https://github.com/googlecartographer/cartographer_ros/issues/264#issuecomment-337916432

closed 2017-10-23 by SirVer