cartographer-project/cartographer #538: about the  coordinate system for cartographer

sunmk2006, 2017-09-21:
hello, How is the coordinate system defined for cartographer?Does this coordinate system correspond to the coordinate system of ROS(x-forward, y-left, z-up) or am I free to define? If it is the latter, how do I define my coordinate system for cartographer?

BrannonKing, 2017-09-22:
Are you asking about Cartographer or the corresponding ROS node?

sunmk2006, 2017-09-25:
yes, i want to know the coordinate system for cartographer,not for the corresponding ros node.

BrannonKing, 2017-09-28:
This question is better for the newsgroup. I've been trying to think how to answer your question. My first thought is right-handed Euclidean. You should be free to define the orientation. In answer to the question "should it correspond to the IMU", the answer is yes, but you may rotate the IMU data before sending it in (just don't translate it). All of the inputs need to be in the same coordinate system (so transform them all to match the IMU).

damonkohler, 2017-09-29:
Cartographer uses the [ROS coordinate system](http://www.ros.org/reps/rep-0103.html). This matters primarily for correctly interpreting range finder data.

closed 2017-09-29 by damonkohler