cartographer-project/cartographer #545: Better support for RGB-D data

cschuet, 2017-09-28:
RGB-D data provides colored point clouds that can be input into Cartographer for scan matching (which would currently ignore  the color information). However the results are quite bad. The current theory is that the noise model of RGB-D cameras is different from that of laser scanners (the depth uncertainty in RGB-D increases steeply with depth, less so for laser scanners).

We should invest more thought into this and find a way to reliably scan match against RGB-D data.

SirVer, 2017-10-04:
Related and possible interesting for tuning: googlecartographer/cartographer_ros#318

Changliu52, 2017-11-03:
I understand the color will be ignored in scan matching, but for a successful rgbd mapping, will the color information remain in the map with the asset_writer?

SirVer, 2017-11-06:
@Changliu52 If it is in the bag, it will also be used in the asset writer.

Changliu52, 2017-11-06:
Perfect! Great. Thank you very much.

Chang

cschuet, 2018-02-23:
Duplicate of #33

closed 2018-02-23 by cschuet