cartographer-project/cartographer #549: Full deserialization from .pbstream task list

ojura, 2017-09-28:
Things to deserialize:

- [x] Submaps (#382)
- [ ] Trajectory nodes (#541)
- [ ] Multiple trajectories (#541)
- [ ] Constraints (#544)
- [ ] IMU data (#548)
- [ ] Odometry data
- [ ] Trajectory connectivity?

Finally:
- [ ] Enable loading unfrozen trajectories (optionally)

ojura, 2017-10-13:
@wohe, I would like to ping you to get your input about the big picture of this.

To summarize, it's a linear series of commits which has been broken into multiple PRs in hope that it would get merged faster, but #554 is basically one big PR (currently with a diff of +377 −71) which contains all of the above (although it does not contain #576, which was my attempt at unblocking #567). The final goal of the last PR, #554, is to enable full unfrozen deserialization, but also to retain the possibility of loading frozen trajectories for pure localization (`LoadMap`, renamed to `LoadState`, accepts a boolean parameter whether to freeze loaded trajectories or not).

Since all these PRs form a continous story of full deserialization, I did not think it would make sense to base them all on top of master, in parallel, and then resolve the unavoidable conflicts -- that's why they all are in a chain. Despite that, you can review the relative changesets from one PR to other by viewing the diffs of their respective branches:

#567: https://github.com/googlecartographer/cartographer/compare/master...larics:loadmap-multi-traj
#544: https://github.com/larics/cartographer/compare/loadmap-multi-traj...spg-deserialization-constraints
#548: https://github.com/larics/cartographer/compare/spg-deserialization-constraints...spg-serialization-imudata
#550: https://github.com/larics/cartographer/compare/spg-serialization-imudata...spg-serialization-odometry
#552: https://github.com/larics/cartographer/compare/spg-serialization-odometry...spg-serialization-misc
#553: https://github.com/larics/cartographer/compare/spg-serialization-misc...spg-serialization-trajectorybuilders
#554: https://github.com/larics/cartographer/compare/spg-serialization-trajectorybuilders...load-unfrozen-state

I would appreciate if you could take the time to review this as a whole. I hoped that we could get this merged soon, since it's currently in a working state and I have a use for it. And keeping up everything with the refactoring currently in progress is exhausting :-( Also, I think it would be beneficial to have this code merged in sooner, so better decisions can be made during the refactoring.

The biggest issue is that #567 currently breaks loading trimmed trajectories, and you deem the remedy in #576 a step back with respect to `MapId` refactoring. Although it is a step back, it would unblock #567 and enable us to merge all of this. Could you reconsider? Or if you have other ideas how to resolve this, I'm open.

wohe, 2017-10-16:
1. I don't think it necessarily makes sense to deserialize trajectory builder options, since we do not plan to continue building a deserialized trajectory. Do you have any particular reason to include it in your list?
2. We plan to add more, e.g. fixed frame pose to 2D, GPS support, landmarks API, so this is a bit of a moving target.
3. If you want to convey a story a better approach would be to open an issue (like this one here), and mention it from all PRs. Keeping them based on top of each other means we have to merge them in order. If you want to merge them faster and have me review some in parallel, keeping them orthogonal and on top of master (should be easily possible, e.g. for supporting odometry and IMU) makes that possible.

ojura, 2017-10-19:
Regarding 1: the trajectory builder options (for example, the map resolution) are used in the realtime assets writer (which was removed from cartographer_ros, but I am still using it).

ojura, 2017-10-19:
@wohe I updated the PRs, PTAL. The first next in line should be #567, which should now be ready.

Regarding your point 3 - I agree that it's better to have them orthogonal, but it really wasn't possible in this case. Most of them modify the same sections of code.

ojura, 2017-10-23:
Ok, next in line is #544.

ojura, 2017-11-02:
Assuming #544 will get merged as it has been approved, next one is #548. I have updated it to follow changes during review of #544.

ojura, 2017-11-09:
#550 is awaiting refactoring of odometry in SPG optimization problem.

#552 is ready.

gaschler, 2018-11-30:
Closing as fixed :-)

closed 2018-11-30 by gaschler