cartographer-project/cartographer #556: Sensor combination for outdoor SLAM

GITSHOHOKU, 2017-09-30:
Hi Guys
   I want to use a combination of RGB-D and Velodyne 16 Lidar for outdoor SLAM project.I don't know if
an IMU is neccesorry to make sure this combination have enough real-time ability and accuracy in practice?
    Many thanks in advance!
Hongkunlong

cschuet, 2017-09-30:
Hi Hongkunlong,

We currently don't have a good story for RGB-D data. The naive approach of converting that to a (uncolored - discarding RGB) point cloud and feeding it into cartographer seems to generate unsatisfactory results (I hear - I haven't tried myself). I filed #545 since RGB-D data is a popular request, but I can't promise when we will have time to address this.

Are you planning to run our 2D or 3D pipeline, i.e. do you want a 2D or 3D map at the end?

Cheers,
Christoph

SirVer, 2017-10-23:
Closing as this issue is not actionable for us.

closed 2017-10-23 by SirVer