cartographer-project/cartographer #557: Cartographer with simulation

Choco83, 2017-10-02:
I am trying to get cartographer running with my gazebo simulation. I have simulate VLP-16 and have tried it with both laserscans and pointcloud, sometimes it seems that cartographer is not generating a map at all. But if I leave rviz and my simulation running I can see map is being built very slowly.
Can you suggest what the problem might be or what parameters I can tweak to get better/quick results. I am running Velodyne at 10hz and 30 degrees fov in simulation.

BrannonKing, 2017-10-02:
Are you using IMU and or odometry data?

Choco83, 2017-10-02:
Yes, I am using both IMU and odom data, simulated IMU and odom I am calculating using wheels' rotation. I am using Kinetic with Gazebo8 and custom model. I am using Velodyne_simulator (pointcloud2) plugin and gazebo's own plugin to switch between pointcloud2 and laserscan.

BrannonKing, 2017-10-02:
Are you using 3D? I assume that you're not. Are you using this configuration? `TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true?`

I'm not suggesting that you use it: it determines what values control performance.

A few things of note: Gazebo's default ODE engine makes very noisy IMU and position values. Bullet works significantly better. Second, odometry is completely broken in the current Cartographer builds. You might try it with just IMU.

Choco83, 2017-10-03:
At the moment I am using 2D, after that I would like to move to 3D.

When I added odometry information it started behaving really weirdly, that's why I guess.
And about the noise, IMU noise was too low so extra noise is added there to make it more realistic.

I will try turning it off, but still, the map generation is too slow. I have to move my model too slow in order to get a meaningful map. Do you know what could be the reason?

SirVer, 2017-10-23:
@Choco83 We are missing a ton of context to provide meaningful help.

If you still require assistance with this, please open a new issue providing everything asked for in the prefilled bug template. Also reference this issue.

closed 2017-10-23 by SirVer

Choco83, 2017-10-24:
@SirVer scans_per_accumulation was the main cause. And at the moment I am not working on simulation. If later I work on it again and find some other/similar issues, I will open new issue and refer to this one. Thanks