cartographer-project/cartographer #560: Trim all submaps if trajectory is pure localization mode

jihoonl, 2017-10-02:
If a trajectory is in a pure localization mode, it should remove all submaps when it finishes.

wohe, 2017-10-05:
I think I'm missing context here. Can you elaborate a bit on what you are trying to achieve? I also do not understand the rationale behind #562.

jihoonl, 2017-10-05:
Assume that you loaded a frozen trajectory and started a pure localization enabled trajectory. Then, robot moved around for a while and the trajectory is finished by calling /finish_trajectory service. The last 3 submaps generated by the pure localization trajectory will remain in the sparse pose graph, and they will be optimized and serialized into pbstream file. 

Since it is _"pure localization"_, I think it should remove trajectory related data when it is finished. Otherwise, the sparse pose graph and pbstream file would get dirtier everytime pure localization trajectory is started and finished.

PR #563 (2017-11-28): Trim all submaps when pure localization trajectory is finished.
Fixes #560 
based on #562
commit 88805a30 (2017-11-28): Trim all submaps when pure localization trajectory is finished. (#563)

Fixes #560 
based on #562
closed 2017-11-28 by wally-the-cartographer