cartographer-project/cartographer #573: Add timing offsets to range data.

wohe, 2017-10-06:
SLAM quality can be improved by applying the transformation most correct in time to each point in range data. For good performance, we do not want to pass each point individually timed through the pipeline, but instead pass range data where each point can have a time offset.

Initial implementation:
1. Add a 4th component to range data representing a time offset in seconds, similar to #571.
2. Handle this data in voxel filtering and accumulation in the local trajectory builders.

Later it might make sense to also change accumulation to be based on time instead of number of range data, or to change the local trajectory builder to take time offsets into account during scan matching (without the need for an unwarping step beforehand). Also, the origin might be interpreted to be at time offset zero, and transformed accordingly for improved quality ray casting.

BrannonKing, 2017-10-06:
So when a laser scan comes in you offset backwards from that message's time stamp, right? Would you use the scan_time parameter in the ROS message to determine that window? That would be helpful. Maybe if it's zero or doesn't exist, you could make an assumption that scans take period * 0.5 time up to 50ms or something.

wohe, 2017-10-06:
This issue is not directly related to ROS, but about adding support for per-point timing to the Cartographer library. One way to make use of it in Cartographer ROS can be seen in googlecartographer/cartographer_ros#526: Using `time_increment`.

PR #601 (2017-10-24): Add time to points in PointCloud.
Adds a type TimedPointCloud that holds 4-dimensional vectors where measurement time of individual points are stored in the fourth entry.
Uses TimedPointCloud for incoming measurements so measurement time is visible to LocalTrajectoryBuilder.

Fixes #573.
commit d4db1e79 (2017-10-24): Add TimedPointCloud and TimedRangeData. (#601)

Adds a type TimedPointCloud that holds 4-dimensional vectors where measurement time of individual points are stored in the fourth entry.
Uses TimedRangeData to pass TimedPointCloud of incoming measurements to LocalTrajectoryBuilder.

Fixes #573.
closed 2017-10-24 by gaschler