cartographer-project/cartographer #579: Cartographer node crash due to IMU

Choco83, 2017-10-12:
Hello!

I am using Velodyne (VLP16) mounted on Jackal to map an environment. I am using Jackal's imu data from a bag file and feeding it to cartographer and some bag files crash giving me following error.

F1012 16:05:27.442504 22674 imu_tracker.cc:39] Check failed: time_ <= time (636432149115879208 vs. 636432149115829239) 
[FATAL] [1507813527.443037166, 1507618111.595044515]: F1012 16:05:27.000000 22674 imu_tracker.cc:39] Check failed: time_ <= time (636432149115879208 vs. 636432149115829239) 
*** Check failure stack trace: ***
    @     0x7fb0357e25cd  google::LogMessage::Fail()
    @     0x7fb0357e4433  google::LogMessage::SendToLog()
    @     0x7fb0357e215b  google::LogMessage::Flush()
    @     0x7fb0357e4e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x545ec4  cartographer::mapping::ImuTracker::Advance()
    @           0x53f68b  cartographer::mapping::PoseExtrapolator::AdvanceImuTracker()
    @           0x53f925  cartographer::mapping::PoseExtrapolator::ExtrapolateRotation()
    @           0x5400c9  cartographer::mapping::PoseExtrapolator::ExtrapolatePose()
    @           0x507590  cartographer_ros::Node::PublishTrajectoryStates()
    @     0x7fb0332d8e3b  ros::TimerManager<>::TimerQueueCallback::call()
    @     0x7fb03330bcf0  ros::CallbackQueue::callOneCB()
    @     0x7fb03330d0f3  ros::CallbackQueue::callAvailable()
    @     0x7fb0333656a1  ros::SingleThreadedSpinner::spin()
    @     0x7fb03334a73b  ros::spin()
    @           0x4fc1bb  cartographer_ros::(anonymous namespace)::Run()
    @           0x4fa714  main
    @     0x7fb031d63830  __libc_start_main
    @           0x4fbda9  _start
    @              (nil)  (unknown)

I thought maybe it was because of some timestamp issues, I made separate node to make sure it doesn't happen and checked all time minimum time difference between 2 consecutive IMU messages, which was also positive, so I guess something else is causing it.

Any takes on this issue. I am attaching my configuration files (no needed i think), and a bag file so you can reproduce this error.

Configuration files:
https://github.com/Choco83/cartographer_config

Bag file:
https://drive.google.com/open?id=0B-B2d5yEVfASMV9iRDJMenBvaDA

I would really appreciate some help in this regard.

SirVer, 2017-10-12:
> ​I thought maybe it was because of some timestamp issues, I made separate node to make sure it doesn't happen and checked all time minimum time difference between 2 consecutive IMU messages, which was also positive, so I guess something else is causing it.

​No, you are exactly right, that is the reason:

~~~
F1012 16:05:27.442504 22674 imu_tracker.cc:39] Check failed: time_ <= time (636432149115879208 vs. 636432149115829239)
[FATAL] [1507813527.443037166, 1507618111.595044515]: F1012 16:05:27.000000 22674 imu_tracker.cc:39] Check failed: time_ <= time (636432149115879208 vs. 636432149115829239)
~~~

My guess is that you checked the timestamp that 'rosbag record' associates with each saved message - this is the received time and basically never useful. What you need to check is the header->stamp of each individual message, i.e. the recording time of the message. But you can save yourself that trouble, because Cartographer does it for you and sees that your messages are out of order. This is unphysical and therefore we crash - simple reordering of your bag using the header timestamp should be sufficient to fix this issue.

​

closed 2017-10-12 by SirVer

Choco83, 2017-10-12:
Thanks for quick respone. I got similar issue with laserscan a while ago and filtered out such messages using a simple node. I used same method here with IMU, and that didn't help so I thought maybe it was something different.
I will try reordering and see if that helps.

SirVer, 2017-10-12:
@Choco83 It could be that this is not your imu, but any of the other sensors you pass to Cartographer. Just make sure that all of your individual frame_ids are time ordered in the bag.

SirVer, 2017-10-12:
Actually, this triggered me to start working on googlecartographer/cartographer_ros#529 which I will send for review tomorrow. It gives the following output when I run it on your test.bag:

~~~
1012 17:10:01.034142 100501 rosbag_wtf_main.cc:91] Sensor with frame_id "imu_link" jumps backwards in time. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
I1012 17:10:01.100076 100501 rosbag_wtf_main.cc:109] Timing histogram for "imu_link":
Count: 1828  Min: -0.004997  Max: 0.032000  Mean: 0.013502
[-0.004997, -0.001297)                          Count: 1 (0.054705%)    Total: 1 (0.054705%)
[-0.001297, 0.002402)                           Count: 0 (0.000000%)    Total: 1 (0.054705%)
[0.002402, 0.006102)                            Count: 0 (0.000000%)    Total: 1 (0.054705%)
[0.006102, 0.009802)                            Count: 3 (0.164114%)    Total: 4 (0.218818%)
[0.009802, 0.013502)        ################    Count: 1419 (77.625824%)        Total: 1423 (77.844643%)
[0.013502, 0.017201)                    ####    Count: 352 (19.256018%) Total: 1775 (97.100655%)
[0.017201, 0.020901)                            Count: 0 (0.000000%)    Total: 1775 (97.100655%)
[0.020901, 0.024601)                            Count: 1 (0.054705%)    Total: 1776 (97.155357%)
[0.024601, 0.028300)                            Count: 0 (0.000000%)    Total: 1776 (97.155357%)
[0.028300, 0.032000]                       #    Count: 52 (2.844639%)   Total: 1828 (100.000000%)
I1012 17:10:01.100162 100501 rosbag_wtf_main.cc:109] Timing histogram for "velodyne":
Count: 243  Min: 0.100746  Max: 0.100941  Mean: 0.100859
[0.100746, 0.100765)                            Count: 2 (0.823045%)    Total: 2 (0.823045%)
[0.100765, 0.100785)                            Count: 1 (0.411523%)    Total: 3 (1.234568%)
[0.100785, 0.100805)                            Count: 0 (0.000000%)    Total: 3 (1.234568%)
[0.100805, 0.100824)                       #    Count: 13 (5.349794%)   Total: 16 (6.584362%)
[0.100824, 0.100844)                      ##    Count: 30 (12.345679%)  Total: 46 (18.930040%)
[0.100844, 0.100863)                 #######    Count: 91 (37.448559%)  Total: 137 (56.378601%)
[0.100863, 0.100883)                 #######    Count: 81 (33.333332%)  Total: 218 (89.711937%)
[0.100883, 0.100902)                       #    Count: 17 (6.995885%)   Total: 235 (96.707817%)
[0.100902, 0.100922)                            Count: 4 (1.646091%)    Total: 239 (98.353912%)
[0.100922, 0.100941]                            Count: 4 (1.646091%)    Total: 243 (100.000000%)
~~~

First, this show that your IMU data is out of order (first line). It also shows that your IMU data comes in with a high variance in timing.

Choco83, 2017-10-13:
Yeah, I also saw this large variance but couldn't detect out of order problem. Thanks for the input.
I will try to figure out why is jackal doing this, because rest of the topics (I filtered those out), are also coming with large variance.