cartographer-project/cartographer #602: SPG Submap serialization - extrapolated poses are serialized the same way as optimized poses

ojura, 2017-10-18:
Submaps for which optimised poses are not available are treated the same way in serialization after `GetSubmapDataUnderLock`  computes an extrapolated pose. During deserialization, in `AddSubmapFromProto`, these extrapolated poses will get injected into `optimized_submap_transforms_` with the rest of submap poses which really _were_ optimized poses. It would be more faithful with respect to the previous state to restore only optimized submap poses, while we should still be able to extrapolate the rest, just like in the original state.

cschuet, 2017-10-23:
What's your specific concern here? It seems to me that storing the extrapolated submap transforms wrongfully in `optimized_submap_transforms_` has neither a user-visible nor an algorithmic impact: after the first run of the global optimization problem Cartographer will be in exactly the same state as if we had remembered which ones were extrapolated.

SirVer, 2017-10-23:
@cschuet Serializing implies to me that writing and loading state is equivalent to just keep the system running. This is not quite correct in this case.

cschuet, 2017-10-23:
@SirVer Ok I see. No, I don't think this is what we are going for, e.g. we are not planning to support continuing to map a deserialized trajectory. I guess what I am asking is: What would you do with that additional information if we stored it?

SirVer, 2017-10-23:
Mhh, true. 

> What would you do with that additional information if we stored it?

What difference does it make right now in the online system if a pose shows up in `optimized_submap_transforms_` or not?

ojura, 2017-10-23:
That `optimized_submap_transforms_` does not influence the algorithm execution might be true, but someone might do something at some point that causes different behaviour to be exhibited. I think that it's a fragile assumption. (And there is some code in `RunOptimization()` which uses `optimized_submap_transforms_` in such a way that I would not put my hand in fire that it works exactly the same after deserialization).

cschuet, 2017-10-23:
> What difference does it make right now in the online system if a pose shows up in `optimized_submap_transforms_` or not?
Seems to me like the main purpose is to be able to provide our best guess at the global submap position for visualization. `SparsePoseGraph::GetAllSubmapData()` is called by cartographer_ros's `MapBuilderBridge`.

Ok we can keep this issue open. So the fix would be to add a `SubmapId` to our proto to remember the last optimized submap and at restoration time only copy those into `optimized_submap_transforms_` that have a SubmapId <= that number.

ojura, 2017-10-30:
I have submitted a PR for this. I am not 100% certain it is correct, but I think it should be a step in the right direction.

wohe, 2017-11-16:
I don't think it makes sense to change this behavior. The only purpose of `optimized_submap_transforms_` is to keep the information about the most recent result of global SLAM, further submap poses are extrapolated. Compare this to trajectory nodes where all global poses (from global SLAM or extrapolated) are stored in the same data structure and the extrapolation is updated as new results are available. We produce global poses as output and the `.pbstream` should contain the SLAM result whether it was in part extrapolated or not.

What seems to be causing the whole discussion to me seems to be bad naming which should be changed to get it into alignment with the terminology documentation: `optimized_submap_transforms_` should be called `global_submap_poses_`. Does that make sense?

ojura, 2017-11-17:
In that case, in order to make everything consistent, would it make sense to insert extrapolated poses when they are requested into (renamed) `global_submap_poses_`?

closed 2017-11-22 by wohe