cartographer-project/cartographer #621: Building error while running catkin_make_isolated --install --use-ninja

shahabty, 2017-11-01:
Hello,
I have been trying to re-install Cartographer. I have removed everything. But, when I tried to install it again I faced this error:
[135/260] Building CXX object CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o
FAILED: /usr/bin/g++ -isystem /usr/include/eigen3 -isystem /home/X/Desktop/catkin_ws/install_isolated/include -isystem /home/X/torch/install/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -I. -I/home/X/Desktop/catkin_ws/src/cartographer -isystem /usr/src/gmock/gtest/include -O3 -DNDEBUG -pthread -std=c++11 -fPIC -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o -MF CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o.d -o CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o -c /home/X/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc
/home/X/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc: In function ‘void cartographer::common::{anonymous}::QuoteStringOnStack(lua_State*)’:
/home/X/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc:49:24: error: ‘lua_pushglobaltable’ was not declared in this scope
lua_pushglobaltable(L); // S: ... string globals
^
/home/X/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc: In lambda function:
/home/X/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc:332:3: error: control reaches end of non-void function [-Werror=return-type]
};
^
cc1plus: some warnings being treated as errors
[135/260] Building CXX object CMakeFiles/cartographer.dir/cartographer/mapping/proto/sparse_pose_graph.pb.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer':
Command '['/home/X/Desktop/catkin_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/X/Desktop/catkin_ws/build_isolated/cartographer && /home/X/Desktop/catkin_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.

Please tell me why I got this error and how to solve this.
Configuration:
ROS lunar
Ubuntu 16.04
Installation guide:
http://wiki.ros.org/lunar/Installation/Ubuntu
https://google-cartographer-ros.readthedocs.io/en/latest/

gaschler, 2017-11-02:
Hi,
seems like (a function of) the Lua library is not found in your build setup.
Please provide more details:
- Which lua library do you have installed?
- `gcc --version`
- Contents of (install|devel)_isolated/cartographer/devel/CMakeCache.txt
- `env | grep ROS`

As always, perhaps you already did this, please retry with a current checkout from git and compile again after deleting all build/install folders.

tentone, 2017-11-02:
Check what directory are being included (its printed on the beginning)

It is probably using an older version of lua that doesn't have the lua_pushglobaltable method.

shahabty, 2017-11-02:
@gaschler 
Lua 5.2.4 
gcc 5.4
However, the lua version that cartographer is using is /usr/lib/x86_64-linux-gnu/liblua5.2.so;/usr/lib/x86_64-linux-gnu/libm.so (found version "5.1.4")
Do you know how to upgrade it to version 5.2?

gaschler, 2017-11-03:
Gcc and Lua versions are fine.
Next step would be to look into actual includes (tentone's suggestion) and CMakeCache.txt

SirVer, 2017-11-06:
@shahabty 

> Lua 5.2.4

This seems untrue. We have two indications that cartographer finds Lua 5.1:

1. `lua_pushglobaltable` was introduced in Lua 5.2 and is not present in 5.1. The compile error you see indicates that Cartographer includes 5.1 headers.
2. The output `Found Lua: /usr/lib/x86_64-linux-gnu/liblua5.2.so;/usr/lib/x86_64-linux-gnu/libm.so (found version "5.1.4")` states that your Lua version is 5.1, even though the file name suggest otherwise. I would guess that cmake found lua 5.1 headers and 5.2. libraries - that will be trouble down the line even if the code compiled.

Our bug here is that we accept 5.1 in the configure step. I opened #629 for that. For your particular instance, I suggest deinstalling 5.1 if that is feasible or hacking `FindLuaGoogle.cmake` to only accept the location of your 5.2. installation.

gaschler, 2017-11-13:
Seems like https://github.com/googlecartographer/cartographer/issues/629 is fixed in general and there is a detailed answer for this particular scenario, so closing.

closed 2017-11-13 by gaschler