cartographer-project/cartographer #622: Header has a std::string using declaration at global scope.

catskul, 2017-11-01:
https://github.com/googlecartographer/cartographer/blob/e10650910e157bd68c55f4d1bc6603f25f89d293/cartographer/common/port.h#L37

Should probably not be in the header file, or at least should be within one of the owned namespaces.

SirVer, 2017-11-02:
Actually that is required, unfortunately. The reason is that inside google, `string` is reimplemented as more efficient solution for our most common use cases, so `std::string` != `string`. It is API compatible though. Since Cartographer has to work in google and outside, we have to use `string` everywhere to be compatible and for the external code, we require this using.

The [abseil](https://github.com/abseil/abseil-cpp) effort is working towards fixing this, so we will be able to soon remove this, but unfortunately not as of right now.

ojura, 2017-11-02:
@SirVer, what exactly is 'google'? :-)

SirVer, 2017-11-02:
@ojura Most of google's source code is stored in [one big repository called Piper](https://cacm.acm.org/magazines/2016/7/204032-why-google-stores-billions-of-lines-of-code-in-a-single-repository/fulltext), this includes compiler, c++ std library and most of googles code. We use [Bazel](https://www.bazel.build/) to build everything from scratch - so, if you write a hello world, we start by compiling the c++ standard library (including our own changes) and end with compiling your source. All dependencies are build from source as well - which is all checked into Piper.

This allows us to make changes like the one I referred to above - but it also means that you require a datacenter to build hello world, more or less.

catskul, 2017-11-03:
@SirVer interesting. 

Would it be possible to move the `using` inside of  the cartographer namespace so that at least it's not in the global namespace? Or is being in the global namespace part of how this string switcheroo trick works?

SirVer, 2017-11-06:
Yes, unfortunately that is the trick. I am investigating if it is feasible to switch to std::string for Cartographer and remove this workaround. It will cost some performance internally, but maybe it doesn't matter as much for Cartographer.

SirVer, 2017-11-06:
> start by compiling the c++ standard library

I was corrected by a colleague that this is not true. We prebuild the standard library, but still build all dependencies of your code.

catskul, 2017-11-06:
Sure.

I'll try to create a patch for this if I can find some time this week

ojura, 2017-11-07:
One other option would be to expose a configurable option through CMake (`CARTOGRAPHER_USE_STD_STRING`?), and in `port.h` wrap the global `using std::string` inside an `#ifdef` which depends on this option.

catskul, 2017-11-07:
Here's one version of a fix, that just moves the typedefs into the cartographer namespace: 

https://github.com/googlecartographer/cartographer/pull/637

I'll add another pull request for the more complete alternative later this week which changes references of string to std::string in the code base.

PR #673 (2017-11-15): replace implicit use of cartographer::string with explicit use of std::string
Fixes #622.
commit f6192e47 (2017-11-15): replace implicit use of cartographer::string with explicit use of std::string (#673)

Fixes #622.
closed 2017-11-15 by wally-the-cartographer