cartographer-project/cartographer #646: Sparse pose graph is a poor name

wohe, 2017-11-09:
The name 'sparse pose graph' is a poor choice since 'pose graph' conveys all the important information and the word 'sparse' is mostly noise: there is no other (dense/non-sparse) pose graph in Cartographer.

The name is used for a subdirectory of the code, for classes `SparsePoseGraph` and in the configuration as `SPARSE_POSE_GRAPH` and `sparse_pose_graph.lua`. Renaming it will break users who have to adapt their configurations.

Unless we can come up with an even better name, I propose dropping the word 'sparse'. Thoughts on this?

ojura, 2017-11-10:
I like the name and how it's a reference to sparse pose adjustment.

```
Pose constraint systems, in typical robotic mapping applications,
exhibit a sparse structure of connections, since the range of
the sensor is typically limited to the vicinity of the robot.
In this paper, we develop a method to efficiently compute the
sparse matrix from the constraint graph, and use direct sparse
linear methods to solve it.
```

If you wanted to convey this, a more accurate name would be e.g. 'sparsely optimised pose graph' or 'SPA pose graph'. But _that_ would be noise :-)

cschuet, 2017-11-10:
I would also argue that since `sparse pose` is a known concept in SLAM it is useful to conserve it in the name.

cschuet, 2017-11-10:
All further discussion on the RFC.

closed 2017-11-10 by cschuet