cartographer-project/cartographer #647: 2D pure localization fails on same map, finds old constraints.

gaschler, 2017-11-10:
As reported by @gladxu in https://github.com/googlecartographer/cartographer_ros/issues/557, 2D pure localization fails on exactly the same map.

```
roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag

roslaunch cartographer_ros demo_backpack_2d_localization.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag map_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag.pbstream
```

[image]

In rviz, one can see that inter constraints are found, but only from the oldest submap of the localization run to the map. Therefore, the reported localization is way off due to the drift in between.

gaschler, 2017-11-10:
In this case, the constraint builder spends more time on scan matching, so work queues up.
Localization on exactly the same map is a special case with a large number of matches.

One way to tune this would be sample fewer potential constraints in general for localization.
I will do more evaluation, but localization has more work with MaybeAddGlobalConstraint than usual SLAM so maybe the default value could be lower
```
SPARSE_POSE_GRAPH.global_sampling_ratio =
    0.2 * SPARSE_POSE_GRAPH.global_sampling_ratio
```

This scenario can be solved with tuning as follows,
```
SPARSE_POSE_GRAPH.global_sampling_ratio = 0.001
SPARSE_POSE_GRAPH.constraint_builder.sampling_ratio = 0.1
```

so closing the issue.

closed 2017-11-10 by gaschler

ojura, 2017-11-11:
Might be useful to add this to the tuning guide.

ojura, 2017-11-23:
@cshuet ping :-)

ojura, 2017-11-23:
@cshuet ping :-) do you agree?

cschuet, 2017-11-23:
Agreed. @gaschler will update the guide.

gaschler, 2018-01-08:
I'm updating the tuning guide with https://github.com/googlecartographer/cartographer_ros/pull/653

sockhands, 2018-01-12:
Is there any way to establish an upper bound to the number of constraints within the same map? It seems that as a robot repeatedly retraces the same path through an established map, it continues to find and add more and more constraints. This corresponds to a gradual increase in CPU load over time, which eventually becomes untenable.

I have tuned various parameters, decreased sampling_ratio, increased the min_score for adding constraints etc. This has helped to slow the increase, but given enough time the constraints inevitably pile up. Does cartographer ever prune/cull older constraints?

gaschler, 2018-01-12:
@sockhands In pure localization mode, constraints of older submaps are pruned (assuming that global SLAM keeps up), even when pruning removes your last constraint to the frozen map. In mapping, constraints will always be added as you expected, but we have successfully run very large problems.

(This question could have been a new issue since is not directly related to this issue.)

sockhands, 2018-01-13:
Thanks @gaschler, I managed to work things out not too long after my comment yesterday. I was confused because the count for added constraints reported by the cartographer log seems to be an ongoing total, and doesn’t drop when an older submap is pruned.