cartographer-project/cartographer #649: Suggestion for the merge bot: merge instead of rebase.

ojura, 2017-11-10:
@SirVer I saw [Wally's logs](https://github.com/googlecartographer/cartographer_ros/pull/561#issuecomment-343440208) and noticed you're rebasing on top of master. Instead of rebase-then-squash, I suggest you merge-then-squash. Some conflicts show up only when rebasing on top of master, and not when merging master. For example:

Suppose that upstream looks like this: `A`
And suppose that the PR looks like this: `A -> C -> revertC -> D`
And suppose that master gets updated: `A -> B`

Commit `B` on `master` causes a conflict with `C`, so `C` cannot be rebased on top of `B`.

However, notice that `D` and `B` can be merged, because D does not contain anything in conflict with `B`, and `C` has been reverted in the PR!

Therefore, I suggest you merge-then-squash.

SirVer, 2017-11-10:
@ojura This belongs into the https://github.com/googlecartographer/infrastructure/ issue tracker, that is where the bot lives.

I can see where you are coming from here. My thought was that rebasing makes reviewing easier in case further manual changes are required or even when the bot adds more commits of its own (clang-format and so on). Your point is valid, but I'd like to wait it out a bit more if it is a nuisance in practice.

closed 2017-11-10 by SirVer

cschuet, 2017-11-10:
Yes, let's discuss this on https://github.com/googlecartographer/infrastructure/pull/1.