cartographer-project/cartographer #657: Cloud mapping

yutakaoka, 2017-11-10:
Hi,

As we talked at the open house, I would like to move forward and contribute something towards cloud mapping.
One of the discussion issue is how to communicate between robots and cloud, or what portion of cartographer should run on the robots or cloud.
I feel like that pure localization should run on the robot due to tight loop, on the back end life long mapping could run on the cloud. One thought is like robots could send insertion_result which local_trajectory_builder makes on the robots to the cloud where global optimization runs, and then at some point, the cloud pushes updated map to the robots.

I wonder if we should just send raw sensor data or internal state like insertion_result via grpc to the cloud.
Any thoughts?

SirVer, 2017-11-10:
Christoph is writing an rfc right now, we will have something to discuss soon I think. @cschuet right?

cschuet, 2017-11-13:
As Holger mentioned I am currently looking into cloud-based mapping designs. I think eventually we will end up with a number of different ways how people go about "cloud-based mapping" as anyone understands something else by that (does the robot run full-blown global SLAM and upstream those "local" views of the global problem?).

With this disclaimer I agree that running pure localization on the agents and upstreaming the local SLAM results to the cloud to solve the global problem is what made most sense to me. So I am currently writing an RFC that describes exactly that. I am hoping to post a high-level strawman RFC before the end of the week.

yutakaoka, 2017-11-13:
That sounds exciting. I look forward to seeing the RFC.
Thanks.
Yutaka

cschuet, 2017-11-16:
You can find a first draft for the RFC [here](https://github.com/googlecartographer/rfcs/pull/7).

cschuet, 2017-11-22:
Closing this issue. Since the RFC is now merged. We will use the corresponding project entry for progress tracking.

closed 2017-11-22 by cschuet

yutakaoka, 2017-11-23:
Thank you for letting me know. 
I am in the Thanksgiving holiday now, so I can't make it to join the open house, but I will catch you up next week.