cartographer-project/cartographer #671: find_package(Protobuf REQUIRED 3.0.0) not working

catskul, 2017-11-15:
Looks like the version argument is in the wrong place. 

I believe it should be:

`find_package(Protobuf 3.0.0 REQUIRED)` 

pull request in a few minutes

catskul, 2017-11-15:
Even once fixed this only seems to work correctly with cmake 3.6+
Checked with cmake 3.5.x and it says:

`-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (Required is at least version "3.0.0") `

even when it is not true.

Cmake 3.6.x correctly reports:

`  Could NOT find Protobuf: Found unsuitable version "2.6.1", but required is
  at least "3.0.0" (found /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread)`

gaschler, 2017-11-15:
Thanks for patching, closing.
Just for documentation: As you said, CMake < 3.6 will not detect the Protobuf version and this will not fix.

closed 2017-11-15 by gaschler