cartographer-project/cartographer #674: catkin_make_isolated --install --use-ninja  FAILED

tseliwang, 2017-11-15:
Hi，
When I execute this line of code ‘catkin_make_isolated --install --use-ninja’, the prompt FAILED. 

cartographer/mapping_3d/proto/submaps_options.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2"

There are seven identical failed

gaschler, 2017-11-15:
Please **install Protobuf >= 3.0.0**.
For most distros, it may be easiest to compile it yourself, following the instructions in https://google-cartographer.readthedocs.io/en/latest/#getting-started-without-ros.
We recently moved to proto3, so this became a dependency.

Background: Only the newest CMake versions can recognize the Protobuf version, so you only learn about this when the (old) protobuf compiler is executed.

closed 2017-11-15 by gaschler

cschuet, 2017-11-15:
You can follow the instructions [here](https://google-cartographer.readthedocs.io/en/latest/).

SirVer, 2017-11-15:
Also: here is the reason why we switch: https://github.com/googlecartographer/rfcs/blob/master/text/0000-proto3-transition.md

tseliwang, 2017-11-15:
In fact, I have installed protobuf 3.4.0 from the website you  provided，but it`s also prompt FAILED.

gaschler, 2017-11-15:
As for all build requirement failures, please provide more details:
- Retry after deleting previous build.
- Look into CMakeCache.txt what requirement it found (in this case, look for protobuf_ variables).
- Post your OS version, CMakeCache.txt and error message here again.
- If you there are multiple versions of the same library, note this here and where they are (this is likely the cause in this case).
- If you think this is related to the problem, specify your compiler, cmake and ros version.

tseliwang, 2017-11-16:
My systeam is Ubuntu 16.04，My ROS is Kinetic. Actually I retry so many times, even if I reinstall Ubuntu, these problem was not be solved. 
I installed it according to the steps of this website. 
https://google-cartographer-ros.readthedocs.io/en/latest/
Here is my CMakeCache.txt and error message. 
[CMakeCache.txt](https://github.com/googlecartographer/cartographer/files/1476947/CMakeCache.txt)
[error.txt](https://github.com/googlecartographer/cartographer/files/1476984/error.txt)

SirVer, 2017-11-16:
Are you sure you executed this line and it succeeded?

~~~
# Install proto3.
src/cartographer/scripts/install_proto3.sh
~~~

tseliwang, 2017-11-16:
Yes, I`m sure I have exectued this line and it worked. When I exectued this `protoc --version` line, The treminal return back `libprotoc 3.4.0`.  I think this problem is multiple versions of the same library, because I knew the ubuntu systeam install one versions of the protobuf when we install the systeam, but I don`t konw how to fix it.

tseliwang, 2017-11-16:
Yes, Im sure I have exectued this line and it worked. When I exectued this
`protoc --version`
line, The treminal return back
`libprotoc 3.4.0. `
I think this problem is multiple versions of the same library, because I knew the ubuntu systeam install one versions of the protobuf when we install the systeam, but I dont konw how to fix it.

cschuet, 2017-11-16:
So when I do
~~~
cd ~/catkin_ws/src/cartographer
rm -Rf build
mkdir build
cd build
cmake ..
~~~

it tells me among other things
~~~
-- Found PROTOBUF: /usr/local/lib/x86_64-linux-gnu/libprotobuf.a
~~~
What do you get?

rykerDial, 2018-03-08:
I was getting this bug with protoc3 installed when trying to build cartographer without installing it: `catkin_make_isolated --use-ninja`.

nwotero, 2019-08-20:
I fixed this issue by running `src/cartographer/scripts/install_proto3.sh` with `sudo` and doing a clean make.

roelofvandijk, 2019-11-07:
Encountered the same problem on Ubuntu 16.04, fixed running the install_proto3.sh script using sudo and doing a clean make. Thanks @nwotero .