cartographer-project/cartographer #690: Cost functions in ceres_scan_matcher

Mofef, 2017-11-20:
We were surprised to see that the scan matching is tied to different poses for rotation and translation each. 
In the translational case the `previous_pose` is used
``` c++
  problem.AddResidualBlock(
      new ceres::AutoDiffCostFunction<TranslationDeltaCostFunctor, 2, 3>(
          new TranslationDeltaCostFunctor(options_.translation_weight(),
                                          previous_pose)),
      nullptr, ceres_pose_estimate);
```
While for rotation the `initial_pose_estimate` (hiding behind `ceres_pose_estimate`) is used.
``` c++
  problem.AddResidualBlock(
      new ceres::AutoDiffCostFunction<RotationDeltaCostFunctor, 1, 3>(
          new RotationDeltaCostFunctor(options_.rotation_weight(),
                                       ceres_pose_estimate[2])),
      nullptr, ceres_pose_estimate);
```
Following the data-flow from `local_trajectory_builder`:
* In case the `real_time_correlative_scan_matcher` is *not* used those two poses hold the same values.
* In case the `real_time_correlative_scan_matcher` *is* used, only the rotation uses the refined pose.

Can you explain what this decision is based on?

wohe, 2017-11-22:
`previous_pose` is due to past usage, it is the estimate from extrapolation now.

The interesting case in 2D SLAM without IMU where the rotation estimate can be pretty bad and the `real_time_correlative_scan_matcher` is used to find a good starting solution. In our experience trusting the orientation of this match is better than the extrapolated orientation estimate since changes in angular velocity can be quite fast. The opposite is true for translations: If the `real_time_correlative_scan_matcher` finds a better match, e.g. from a repetitive pattern, it is a bad idea to not penalize the change in velocity. If we would use the result from the `real_time_correlative_scan_matcher` here, huge changes in velocity could be accepted which break SLAM.

closed 2017-11-23 by cschuet

Mofef, 2017-11-23:
Great, thanks! That makes a lot of sense to me now.