cartographer-project/cartographer #705: offline_backpack_3d Dropped empty range data

chenyang-08s, 2017-11-24:
Hi, I want to use the offline_backpack_3d example to generate a 3D map from a bag file. 
Because our robot is not built on ROS, I converted the data from time stamped pcd files and IMU outputs to a Rosbag file using the Rosbag API.
I tried LOAM before and the data could be displayed on RVIZ correctly. Our robot has a single VLP16 Lidar and a IMU on board.
I have edited the configuration files accordingly but nothing is showing up in RVIZ and local_trajectory_builder.cc throws warnings about empty range data and the trajectory seems to finish once it is generated. 
Could you please take a look and help to point out where I might have missed? All relevant files can be found in the link below, if you need any additional info please let me know. Thanks!

[https://github.com/chenyang-08s/cartographer_files/blob/master/issue_template.md](url)

gaschler, 2017-11-27:
Let's first run our shiny new diagnosis tool `cartographer_rosbag_validate` 
```
./devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_rosbag_validate  -bag_filename=$HOME/Downloads/test_imu_cart.bag 
I1127 10:41:52.366327 153154 rosbag_validate_main.cc:155] Time delta histogram for consecutive messages on topic "imu/data" (frame_id: "imu_link"):
Count: 2297  Min: 0.007123  Max: 0.712521  Mean: 0.031876
[0.007123, 0.077663)	 ###################	Count: 2170 (94.471046%)	Total: 2170 (94.471046%)
[0.077663, 0.148203)	                   #	Count: 117 (5.093600%)	Total: 2287 (99.564651%)
[0.148203, 0.218742)	                    	Count: 8 (0.348280%)	Total: 2295 (99.912933%)
[0.218742, 0.289282)	                    	Count: 0 (0.000000%)	Total: 2295 (99.912933%)
[0.289282, 0.359822)	                    	Count: 0 (0.000000%)	Total: 2295 (99.912933%)
[0.359822, 0.430362)	                    	Count: 1 (0.043535%)	Total: 2296 (99.956467%)
[0.430362, 0.500902)	                    	Count: 0 (0.000000%)	Total: 2296 (99.956467%)
[0.500902, 0.571442)	                    	Count: 0 (0.000000%)	Total: 2296 (99.956467%)
[0.571442, 0.641981)	                    	Count: 0 (0.000000%)	Total: 2296 (99.956467%)
[0.641981, 0.712521]	                    	Count: 1 (0.043535%)	Total: 2297 (100.000000%)
I1127 10:41:52.366879 153154 rosbag_validate_main.cc:155] Time delta histogram for consecutive messages on topic "velodyne_points" (frame_id: "vlp16_link"):
Count: 729  Min: 0.095829  Max: 0.104169  Mean: 0.100000
[0.095829, 0.096663)	                    	Count: 1 (0.137174%)	Total: 1 (0.137174%)
[0.096663, 0.097497)	                    	Count: 0 (0.000000%)	Total: 1 (0.137174%)
[0.097497, 0.098331)	                    	Count: 0 (0.000000%)	Total: 1 (0.137174%)
[0.098331, 0.099165)	                    	Count: 5 (0.685871%)	Total: 6 (0.823045%)
[0.099165, 0.099999)	              ######	Count: 227 (31.138546%)	Total: 233 (31.961592%)
[0.099999, 0.100833)	       #############	Count: 490 (67.215363%)	Total: 723 (99.176956%)
[0.100833, 0.101667)	                    	Count: 5 (0.685871%)	Total: 728 (99.862823%)
[0.101667, 0.102501)	                    	Count: 0 (0.000000%)	Total: 728 (99.862823%)
[0.102501, 0.103335)	                    	Count: 0 (0.000000%)	Total: 728 (99.862823%)
[0.103335, 0.104169]	                    	Count: 1 (0.137174%)	Total: 729 (100.000000%)
```
The only thing that here is that frequencies could be higher. Does your IMU sensor provide only one measurement every 30ms? Also, it has a high jitter.

Inspecting your bag with rqt_bag shows that your IMU linear_acceleration is roughly [0, 0, -1]. However, we expect it in the format m/s^2, so rather [0, 0, 9.81]. This must be fixed.

Running `roslaunch cartographer_ros demo_kaibo.launch`  shows a warning that no IMU data is coming in. The 3D pipeline of cartographer requires IMU data.

[ WARN] [1511775107.592779250, 1511486571.341885563]: W1127 10:31:47.000000 139228 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)

What you need is to remap to the correct topic name by adding
`<remap from="imu" to="imu/data" />
`
to cartographer_ros/launch/kaibo.launch.

Also, because your point cloud input has a low frequency, we need to adapt the configuration to accumulate fewer scans before scan matching.
`-TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160
+TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 5
 
 MAP_BUILDER.use_trajectory_builder_3d = true
 MAP_BUILDER.num_background_threads = 7
 POSE_GRAPH.optimization_problem.huber_scale = 5e2
-POSE_GRAPH.optimize_every_n_nodes = 320
`
This works barely, but as I said, the IMU data format must be fixed.

Thanks for providing all the configuration to make your issue reproducible. In the future, it would be even easier to reproduce if you provide a fork of cartographer_ros that contains exactly your configuration.

chenyang-08s, 2017-11-29:
Hi @gaschler thank you for your time.

I think I have linked the older version of the rosbag. I did fix the remap problem before opening this issue.

So for me, I did not see the same warning as you did. Cartographer was not complaining about not being able to receive imu data. 

In my case, I keep seeing this "Dropped empty range data." warning even after I fixed the linear acceleration units to match the m/s^2 format.

Also, as I changed 
``` bash
TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 5
```
as you suggested, the "Dropped empty range data." warning just appeared more often than before. The map frame is still missing in RVIZ (I'm guessing SLAM is still not working?)

Here are the links to the fixed rosbag and terminal output after running
``` bash
roslaunch cartographer_ros demo_kaibo.launch
```

[kaibo.bag](https://drive.google.com/open?id=1pmT-G3oVh3PxRMrI2EBo6uhdIvaBBuCW)
[output](https://gist.github.com/chenyang-08s/addd96770568f8b52b71a44c2659cef8)
[config files in my fork](https://github.com/chenyang-08s/cartographer_ros)

I was suspecting my cartographer building or installation but I remembered that I was able to run the 3D backpack demo without problem.
``` bash
roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=${HOME}/Downloads/b3-2016-04-05-14-14-00.bag
```
 Please point out if you spot anything else. Thanks again!

chenyang-08s, 2017-11-29:
Hi @gaschler 

And I reinstalled cartographer anyway. The dropped empty data error disappeared and this output was found instead, similar output with the same error message is found when running cartographer in a docker container.
``` txt
$ roslaunch cartographer_ros demo_kaibo.launch 
... logging to /home/chen/.ros/log/adfea54e-d4d8-11e7-8999-7085c2452675/roslaunch-chen-desktop-28162.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://chen-desktop:34365/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [28173]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to adfea54e-d4d8-11e7-8999-7085c2452675
process[rosout-1]: started with pid [28186]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [28193]
process[cartographer_node-3]: started with pid [28208]
process[cartographer_occupancy_grid_node-4]: started with pid [28212]
process[rviz-5]: started with pid [28232]
process[playbag-6]: started with pid [28256]
[ INFO] [1511941275.808764364]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_3d_kaibo.lua' for 'backpack_3d_kaibo.lua'.
[ INFO] [1511941275.808986534]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1511941275.809012311]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1511941275.809049208]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1511941275.809070594]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1511941275.809176792]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1511941275.809199982]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1511941275.809236921]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1511941275.809257554]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1511941275.809319070]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1511941275.809342648]: I1129 16:41:15.000000 28208 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1511941275.813636934]: I1129 16:41:15.000000 28208 submaps.cc:295] Added submap 1
[ INFO] [1511941275.813672086]: I1129 16:41:15.000000 28208 map_builder_bridge.cc:101] Added trajectory with ID '0'.
0x1b1a790 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1075340) ): Attempt to set a screen on a child window.
0x1b1a080 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1075340) ): Attempt to set a screen on a child window.
0x1b224e0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1075340) ): Attempt to set a screen on a child window.
0x1b16a70 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1075340) ): Attempt to set a screen on a child window.
[ INFO] [1511941276.711228143, 1511914930.582696644]: I1129 16:41:16.000000 28208 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636475117305847109'.
F1129 16:41:17.535328 28208 imu_tracker.cc:66] Check failed: (orientation_ * gravity_vector_).z() > 0. (-nan vs. 0) 
[FATAL] [1511941277.536013686, 1511914930.996466332]: F1129 16:41:17.000000 28208 imu_tracker.cc:66] Check failed: (orientation_ * gravity_vector_).z() > 0. (-nan vs. 0) 
*** Check failure stack trace: ***
    @     0x7ff003f575cd  google::LogMessage::Fail()
    @     0x7ff003f59433  google::LogMessage::SendToLog()
[... cut ...] _name:=cartographer_node __log:=/home/chen/.ros/log/adfea54e-d4d8-11e7-8999-7085c2452675/cartographer_node-3.log].
log file: /home/chen/.ros/log/adfea54e-d4d8-11e7-8999-7085c2452675/cartographer_node-3*.log
^C[playbag-6] killing on exit
[rviz-5] killing on exit
[cartographer_occupancy_grid_node-4] killing on exit
[robot_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

gaschler, 2017-11-29:
The failed check shows that `orientation_ * gravity_vector_` is NAN.
Please ensure that your IMU data contains no NAN values.
Either by inspecting the first few messages manually with rqt_bag or by adding more checks and log output inside LocalTrajectoryBuilder::AddImuData or both.
Also, please post the output of cartographer_rosbag_validate for your correct bag file.
Thanks.

ojura, 2017-11-29:
> The failed check shows that orientation_ * gravity_vector_ is NAN.
Please ensure that your IMU data contains no NAN values.

Actually, I managed to run into this exact check failing, while curiously having no IMU data at all (only odometry). Although I haven't checked odometry for NaNs. I will investigate/submit an issue next week.

gaschler, 2017-11-29:
@ojura Very interesting, would be great to have a (small) reproducible case.

chenyang-08s, 2017-11-30:
@gaschler Thanks! That was the first thing I checked. I used rostopic echo to check the first part of the IMU data and roughly went through the bag but could not find any NAN values. I just used rqt_bag again but couldn't find any either. I will try to add logs to see where it actually complains.

Output of cartographer_rosbag_validate for correct bag file:
``` txt
rosrun cartographer_ros cartographer_rosbag_validate -bag_filename=/home/chen/Downloads/kaibo.bag 
I1130 09:33:11.125221 10459 rosbag_validate_main.cc:155] Time delta histogram for consecutive messages on topic "imu" (frame_id: "imu_link"):
Count: 2297  Min: 0.007123  Max: 0.712521  Mean: 0.031876
[0.007123, 0.077663)	 ###################	Count: 2170 (94.471046%)	Total: 2170 (94.471046%)
[0.077663, 0.148203)	                   #	Count: 117 (5.093600%)	Total: 2287 (99.564651%)
[0.148203, 0.218742)	                    	Count: 8 (0.348280%)	Total: 2295 (99.912933%)
[0.218742, 0.289282)	                    	Count: 0 (0.000000%)	Total: 2295 (99.912933%)
[0.289282, 0.359822)	                    	Count: 0 (0.000000%)	Total: 2295 (99.912933%)
[0.359822, 0.430362)	                    	Count: 1 (0.043535%)	Total: 2296 (99.956467%)
[0.430362, 0.500902)	                    	Count: 0 (0.000000%)	Total: 2296 (99.956467%)
[0.500902, 0.571442)	                    	Count: 0 (0.000000%)	Total: 2296 (99.956467%)
[0.571442, 0.641981)	                    	Count: 0 (0.000000%)	Total: 2296 (99.956467%)
[0.641981, 0.712521]	                    	Count: 1 (0.043535%)	Total: 2297 (100.000000%)
I1130 09:33:11.125843 10459 rosbag_validate_main.cc:155] Time delta histogram for consecutive messages on topic "velodyne_points" (frame_id: "vlp16_link"):
Count: 729  Min: 0.095829  Max: 0.104169  Mean: 0.100000
[0.095829, 0.096663)	                    	Count: 1 (0.137174%)	Total: 1 (0.137174%)
[0.096663, 0.097497)	                    	Count: 0 (0.000000%)	Total: 1 (0.137174%)
[0.097497, 0.098331)	                    	Count: 0 (0.000000%)	Total: 1 (0.137174%)
[0.098331, 0.099165)	                    	Count: 5 (0.685871%)	Total: 6 (0.823045%)
[0.099165, 0.099999)	              ######	Count: 227 (31.138546%)	Total: 233 (31.961592%)
[0.099999, 0.100833)	       #############	Count: 490 (67.215363%)	Total: 723 (99.176956%)
[0.100833, 0.101667)	                    	Count: 5 (0.685871%)	Total: 728 (99.862823%)
[0.101667, 0.102501)	                    	Count: 0 (0.000000%)	Total: 728 (99.862823%)
[0.102501, 0.103335)	                    	Count: 0 (0.000000%)	Total: 728 (99.862823%)
[0.103335, 0.104169]	                    	Count: 1 (0.137174%)	Total: 729 (100.000000%)
```

chenyang-08s, 2017-11-30:
@gaschler So I was trying to print and see which data was causing the NAN and I found out that a empty IMU data with linear acceleration of 0,0,0 was pushed into the imu_data_ array before the first imu reading in my bag file. I tried to skip that specific empty imu reading or assign a dummy value to the linear acceleration in the z axis but both ways resulted in a SEGFAULT instead. Having no success on finding out where the problem originated from, I tried to run the 3D backpack demo example again and the error is still there. After that I rebuilt cartographer in another workspace directory and got the same error:

``` txt

roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=/home/chen/Downloads/b3-2016-04-05-14-14-00.bag 
... logging to /home/chen/.ros/log/c8041794-d5aa-11e7-acf0-7085c2452675/roslaunch-chen-desktop-3678.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://chen-desktop:34551/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [3702]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c8041794-d5aa-11e7-acf0-7085c2452675
process[rosout-1]: started with pid [3715]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [3718]
process[cartographer_node-3]: started with pid [3719]
process[cartographer_occupancy_grid_node-4]: started with pid [3720]
process[rviz-5]: started with pid [3721]
process[playbag-6]: started with pid [3722]
[ INFO] [1512031516.634297054]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/src/cartographer_ros/cartographer_ros/configuration_files/backpack_3d.lua' for 'backpack_3d.lua'.
[ INFO] [1512031516.717093296]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1512031516.717273842]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1512031516.751491526]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1512031516.751665280]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1512031516.752419149]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1512031516.752541824]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1512031516.758369461]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1512031516.758508204]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1512031516.759847069]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1512031516.759983977]: I1130 17:45:16.000000  3719 configuration_file_resolver.cc:41] Found '/home/chen/Documents/catkin_ws_test/devel/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1512031516.842436255]: I1130 17:45:16.000000  3719 submaps.cc:295] Added submap 1
[ INFO] [1512031516.842580386]: I1130 17:45:16.000000  3719 map_builder_bridge.cc:101] Added trajectory with ID '0'.
0x162f330 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0xb88390) ): Attempt to set a screen on a child window.
0x162eb70 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0xb88390) ): Attempt to set a screen on a child window.
0x162b4e0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0xb88390) ): Attempt to set a screen on a child window.
0x1637ad0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0xb88390) ): Attempt to set a screen on a child window.
[ INFO] [1512031576.505355672, 1459865640.756439518]: I1130 17:46:16.000000  3719 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '635954624407449496'.
[... cut ...]
[rviz-5] killing on exit
[cartographer_occupancy_grid_node-4] killing on exit
[robot_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
Using GDB to run the cartographer node shows that the program segfaulted at line 177 of src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc:

```cc

    submap_entry.submap_version = submap_id_data.data.submap->num_range_data();

```

closed 2017-12-06 by chenyang-08s