cartographer-project/cartographer #722: Queue waiting for data: (0, points2)

nitramek, 2017-11-30:
Hi,

My setup (right now) is single VLP16 + Android phone as IMU (to be replaced, now just for testing). I am trying to make cartographer work and then try to calibrate settings and such. Also running on Ubuntu 16.04 LTS, ROS Kinetic. In online mode I run velodyne driver and each UDP packet is send as PointCloud2 as suggested.

My goal is to get 3D SLAM up and running but right now it gives me this warning and cartographer is producing no output. I have tried to debug the pipeline with some print statements but with no sucess. I found out the data is coming through but thats it.

The exact configuration lies in forked repo in the nitramek branch.
[https://github.com/nitramek/cartographer_ros/tree/nitramek/cartographer_ros](https://github.com/nitramek/cartographer_ros/tree/nitramek/cartographer_ros)
Direct links:
- [Launch file](https://github.com/nitramek/cartographer_ros/blob/nitramek/cartographer_ros/launch/cart_test.launch)
- [URDF](https://github.com/nitramek/cartographer_ros/blob/nitramek/cartographer_ros/urdf/cart_test.urdf) - doesnt reflect the setup exactly the IMU is really close to the VLP sensor. Right now I am trying to just have some output. 
- [LUA config](https://github.com/nitramek/cartographer_ros/blob/nitramek/cartographer_ros/configuration_files/cart_test.lua) - I have read that if the data from the VLP is divided in each packet the num_accumulated_range_data should be higher but I tried that configuration too and it didnt help.

Output of roslaunch with cart_test.bag
```roslaunch cartographer_ros cart_test.launch bag_filename:=/home/nitramek/vsb/bags/cart_test.bag 
... logging to /home/nitramek/.ros/log/554263dc-d60e-11e7-8959-000c29996d44/roslaunch-ubuntu-22028.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:41761/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [22039]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 554263dc-d60e-11e7-8959-000c29996d44
process[rosout-1]: started with pid [22052]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [22056]
process[cartographer_node-3]: started with pid [22061]
process[cartographer_occupancy_grid_node-4]: started with pid [22063]
process[rviz-5]: started with pid [22067]
process[playbag-6]: started with pid [22081]
[ INFO] [1512074271.288926372]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer_ros/configuration_files/cart_test.lua' for 'cart_test.lua'.
[ INFO] [1512074271.290025782]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1512074271.290178792]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1512074271.290376293]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1512074271.290455295]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1512074271.290619226]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1512074271.290729493]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1512074271.290861462]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1512074271.290935842]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1512074271.291261740]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1512074271.291402077]: I1130 21:37:51.000000 22061 configuration_file_resolver.cc:41] Found '/home/nitramek/ros_all/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1512074271.332676621]: I1130 21:37:51.000000 22061 submaps.cc:295] Added submap 1
[ INFO] [1512074271.332857002]: I1130 21:37:51.000000 22061 map_builder_bridge.cc:101] Added trajectory with ID '0'.
[ INFO] [1512074271.736876989, 1511821305.741212741]: I1130 21:37:51.000000 22061 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636475895586160000'.
[ WARN] [1512074277.194455759, 1511821311.204946129]: W1130 21:37:57.000000 22061 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1512074277.227861131, 1511821311.236593011]: W1130 21:37:57.000000 22061 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1512074277.302636897, 1511821311.311198297]: W1130 21:37:57.000000 22061 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1512074277.382513099, 1511821311.395140797]: W1130 21:37:57.000000 22061 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1512074277.464577494, 1511821311.478894225]: W1130 21:37:57.000000 22061 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[... 147 lines cut ...]
[0.000000, 0.002200)	                ####	Count: 186 (19.558359%)	Total: 186 (19.558359%)
[0.002200, 0.004400)	                    	Count: 0 (0.000000%)	Total: 186 (19.558359%)
[0.004400, 0.006600)	                    	Count: 0 (0.000000%)	Total: 186 (19.558359%)
[0.006600, 0.008800)	                    	Count: 23 (2.418507%)	Total: 209 (21.976866%)
[0.008800, 0.011000)	             #######	Count: 348 (36.593060%)	Total: 557 (58.569927%)
[0.011000, 0.013200)	                ####	Count: 188 (19.768665%)	Total: 745 (78.338593%)
[0.013200, 0.015400)	                    	Count: 0 (0.000000%)	Total: 745 (78.338593%)
[0.015400, 0.017600)	                    	Count: 1 (0.105152%)	Total: 746 (78.443741%)
[0.017600, 0.019800)	                  ##	Count: 96 (10.094637%)	Total: 842 (88.538383%)
[0.019800, 0.022000]	                  ##	Count: 109 (11.461619%)	Total: 951 (100.000000%)
I1130 21:27:53.165081 21421 rosbag_validate_main.cc:155] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
Count: 7244  Min: 0.000091  Max: 0.290574  Mean: 0.001374
[0.000091, 0.029139)	####################	Count: 7243 (99.986198%)	Total: 7243 (99.986198%)
[0.029139, 0.058188)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.058188, 0.087236)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.087236, 0.116284)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.116284, 0.145333)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.145333, 0.174381)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.174381, 0.203429)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.203429, 0.232477)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.232477, 0.261526)	                    	Count: 0 (0.000000%)	Total: 7243 (99.986198%)
[0.261526, 0.2905
```

Last thing there is no movement of the test cart with VLP sensor in this bag too, it is just for example purposes right now.

gaschler, 2017-12-01:
Hi,
first of all, thanks for providing a very well documented issue.

I downloaded cart_test_fixed.bag and inspected it with `rqt_bag ~/Downloads/cart_test_fixed.bag`.
The two sensors have header time stamps that differ by 100000 seconds.
(Look at `header.stamp.secs`)
Velodyne points are so far behind they are never dispatched.
Cartographer requires good synchronization (<20ms) between points and IMU.
This must be fixed.
Perhaps using a separate device (phone) as an IMU is not good enough for even a first test, not sure if you will be able to synchronize it well enough.

Also, there are gaps larger than 0.1 s in IMU and 0.5 s in points data, which is bad.
The topic names and IMU data look fine to me.

nitramek, 2017-12-01:
Hello Gashchler, thanks for quick feedback! I only glanced at those stamps didnt see it, will check/fix the bag and report back.

nitramek, 2017-12-02:
UPDATE:
I am using this android implementation https://github.com/rpng/android_sensors_driver and I have experimented with [Line 230 IMU publisher](https://github.com/rpng/android_sensors_driver/blob/master/sensors_driver/src/udel/rpng/sensors_driver/publishers/ImuPublisher.java#L230).

When I assigned the time to be just `System.currentTimeInMillis()`, I was able to see output through Cartographer. 

Thanks again.

I still have question regarding the IMU and point cloud, how fast the data rates should be? I have looked at the example 3D muzeum bag and the IMU is quite bit slower the PointCloud2s. Also what are you using to synchronize the IMU and PointCloud2 messages or where I can read about it?

closed 2017-12-02 by nitramek