cartographer-project/cartographer #743: pure_location Loadmap time cost  6 second?

xuguangyun, 2017-12-07:
I have compile the Cartographer with Debug and run it. But there have some issue:  Loadmap function cost about 6 second (the map size 19M)  and terminal out isnot normal,as this:
[image]

gaschler, 2017-12-07:
Yes, debug mode is really inefficient. The terminal output tells the same.
If you need debug symbols, please compile with CMAKE_BUILD_TYPE=RelWithDebInfo.
Closing because it works as intended.

closed 2017-12-07 by gaschler

xuguangyun, 2017-12-08:
Hi gaschler,thanks a lot.As you say i compile with CMAKE_BUILD_TYPE=RelWithDebInfo but it turn out to be  the same ..

xuguangyun, 2017-12-08:
solved.Same like gaschler say.but i donot know why the CMAKE_BUILD_TYPE=RelWithDebInfo do not work.
i have to use add the command   -std=gnu++11  -O3 .