cartographer-project/cartographer #765: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty

will1991, 2017-12-16:


cschuet, 2017-12-16:
Can you provide additional information? What happened? What were you trying to do?

DennisBian, 2017-12-29:
The long corridor problem makes it really difficult to get the correct pose without using odometry ,  slam think the robot is not moving when laser scan datas are almost the same at long corridor, thus i think using odometry might sloves the problem.  However, I have met the same issue when i set
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry = false,

ojura, 2017-12-29:
The sensor messages you are passing to Cartographer probably have an empty frame id, which is invalid.

MichaelGrupp, 2018-01-03:
@DennisBian `provide_odom_frame` does not mean that you use a separate odometry source, that's the job of `use_odometry`. See also [here](https://github.com/googlecartographer/cartographer_ros/blob/master/docs/source/configuration.rst). If you still have problems, I'd recommend to open a separate issue since this "long corridor" problem is not really related to the original post here.

DennisBian, 2018-01-03:
@MichaelGrupp sorry for my mistake typing, i just setting the "use_odometry = true,"  and when i changed the odom topic name as remap from "odometry" ,i have fixed this problem.

gaschler, 2018-01-03:
Closing this as answered. Feel free to re-open if needed.

closed 2018-01-03 by gaschler