cartographer-project/cartographer #789: Interface to trim arbitrary submaps

yutakaoka, 2018-01-05:
Is there any way to trim arbitrary submaps? #409 seems to ready to go, but as far as I know, there is only one implementation([PureLocalizationTrimmer](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/pose_graph_trimmer.h#L58)) for pure localization. What I want to do is trimming submaps which can be identified with SubmapId during mapping.
Any info or direction for that would be appreciated.

gaschler, 2018-01-24:
You can trim arbitrary submaps by id, it is supported since ~Nov 2017.
As you said, there is currently no example other than PureLocalizationTrimmer.
But you can go ahead an implement your own Trimmer inheriting from `PoseGraphTrimmer`.

cschuet, 2018-01-29:
Closing as answered.

closed 2018-01-29 by cschuet