cartographer-project/cartographer #806: Get initial trajectory pose from .lua files

EdwardLiuyc, 2018-01-10:
I want to set an initial pose to cartographer, but it seems that the option `initial_trajectory_pose` is not read in function`proto::TrajectoryBuilderOptions CreateTrajectoryBuilderOptions`, or is there another way to set the initial pose.

EdwardLiuyc, 2018-01-13:
I have set initial pose "manually" with code like this, is that suitable doing this way? and i want know if the function of setting initial pose tested. 
```
::cartographer::mapping::proto::InitialTrajectoryPose initial_trajectory_pose;
	::cartographer::transform::proto::Rigid3d 		relative_pose;
	::cartographer::transform::proto::Quaterniond 	rotation;
	::cartographer::transform::proto::Vector3d		translation;
	translation.set_x( 0. );
	translation.set_y( 0. );
	translation.set_z( 0. );
        rotation.set_w( 1. );
        rotation.set_x( 0. );
        rotation.set_y( 0. );
	rotation.set_z( 0. );
	initial_trajectory_pose.set_to_trajectory_id( 0 );
	*relative_pose.mutable_rotation() = rotation;
	*relative_pose.mutable_translation() = translation;
	*initial_trajectory_pose.mutable_relative_pose() = relative_pose;
	*(trajectory_options->trajectory_builder_options.mutable_initial_trajectory_pose()) = initial_trajectory_pose;
```

cschuet, 2018-01-24:
@EdwardLiuyc We confirmed that the feature works. Note that you need to specify the initial pose relative to an existing trajectory. Currently an initial pose can be set through a call to `cartographer_start_trajectory`  by passing the `-initial_pose` flag. There is an [RFC](https://github.com/googlecartographer/rfcs/pull/17) pending that proposes to use the `/initialpose` topic.

closed 2018-01-24 by cschuet

NTUwilliam, 2019-03-01:
@cschuet 
> need to specify the initial pose relative to an existing trajectory

What do you mean by this? Is it to specify the **tf** from _new initial pose_ to the _previous initial pose in the pre-mapping phase_?

> an initial pose can be set through a call to `cartographer_start_trajectory` by passing the `-initial_pose` flag

I've browsed through the [RFC](https://github.com/googlecartographer/rfcs/pull/17) but still not clear how to set the `initial_trajectory_pose` :( Could you give more details about this? Like which part of the code I need to modify.

Any hint or advice is appreciated. Thanks a lot!