cartographer-project/cartographer #812: How to set up the environment variables of two spaces

ytgcljj, 2018-01-12:
I have two workspaces, onecartographer, one velodyne, cartographer is compiled by catkin_make_iso, velodyn is compiled by catkin_make, now the question is how to set up two environment variables, can I use an environment variable?
 thanks

SirVer, 2018-01-12:
You can't, but you can just `mv first_space/src/* second_space/src` and only have one space.

this is also not related to Cartographer. Should the above suggestion not work for you, please try https://answers.ros.org/questions/.

closed 2018-01-12 by SirVer

ytgcljj, 2018-01-12:
I am under a SRC, but these are two variables, one is under devel, one is under install_isolated, I cannot install cartographer, no other space can be installed.

SirVer, 2018-01-12:
just move the velodyne one to the cartographer workspace and build both with `catkin_make_isolated`. I did that before, it works.