cartographer-project/cartographer #823: Empty range_data causes fatal error

ryanmeasel, 2018-01-16:
I collected a ROS bag for a unit with 2 VLPs and an IMU using `cartographer_ros` and cartographer was able to run for the full duration of the scan. When the bag is processed however, a fatal error occurs:

```bash
[FATAL] [1516059899.205516676, 1515519557.762312447]: F0115 15:44:59.000000 12672 local_trajectory_builder.cc:71] Check failed: !range_data.returns.empty() 
```
The error is correct as the bag does contain a LiDAR packet with an empty data payload near that timestamp. Does this need to be fatal? It seems like the packet could simply be discarded. Also, why would cartographer continue to run without error during collection but not during playback?

I have enclosed the full run output as well as the bag validation:
```bash
$ roslaunch cartographer_ros offline_backpack_2vlp.launch bag_filenames:=/home/ryan/Downloads/cartographer_2018-01-09-09-38-35.bag 
... logging to /home/ryan/.ros/log/0570f20a-fa4a-11e7-9e23-408d5c0720fc/roslaunch-snorlax-12654.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://snorlax:46769/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /use_sim_time: True

NODES
  /
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    cartographer_offline_node (cartographer_ros/cartographer_offline_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rviz-1]: started with pid [12671]
process[cartographer_offline_node-2]: started with pid [12672]
process[cartographer_occupancy_grid_node-3]: started with pid [12673]
[ INFO] [1516059876.933323027]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/mapping_backpack_2vlp.lua' for 'mapping_backpack_2vlp.lua'.
[ INFO] [1516059876.933592198]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1516059876.933623122]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1516059876.933673764]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1516059876.933699973]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1516059876.933818685]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1516059876.933847147]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1516059876.933889700]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1516059876.933909724]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1516059876.933986135]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1516059876.934009268]: I0115 15:44:36.000000 12672 configuration_file_resolver.cc:41] Found '/home/ryan/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1516059876.941930782]: I0115 15:44:36.000000 12672 submaps.cc:295] Added submap 1
[ INFO] [1516059876.942006980]: I0115 15:44:36.000000 12672 map_builder_bridge.cc:90] Added trajectory with ID '0'.
[ INFO] [1516059883.537822137, 1515519515.758172890]: I0115 15:44:43.000000 12672 offline_node.cc:183] Processed 0 of 696.342 bag time seconds...
[ INFO] [1516059883.538446351, 1515519515.766045467]: I0115 15:44:43.000000 12672 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636511163157662360'.
[ INFO] [1516059883.538633079, 1515519515.768428106]: I0115 15:44:43.000000 12672 local_trajectory_builder.cc:67] IMU not yet initialized.
[ INFO] [1516059883.538751940, 1515519515.768444131]: I0115 15:44:43.000000 12672 local_trajectory_builder.cc:67] IMU not yet initialized.
[ INFO] [1516059891.949071815, 1515519542.140733122]: I0115 15:44:51.000000 12672 collated_trajectory_builder.cc:69] /horiz_lidar_points rate: 1492.86 Hz 6.70e-04 s +/- 5.21e-04 s (pulsed at 313.58% real time)
[ INFO] [1516059891.949156152, 1515519542.140733122]: I0115 15:44:51.000000 12672 collated_trajectory_builder.cc:69] /imu/data rate: 99.98 Hz 1.00e-02 s +/- 1.29e-03 s (pulsed at 313.98% real time)
[ INFO] [1516059891.949574730, 1515519542.140733122]: I0115 15:44:51.000000 12672 collated_trajectory_builder.cc:69] /vert_lidar_points rate: 1492.13 Hz 6.70e-04 s +/- 5.66e-04 s (pulsed at 314.03% real time)
[ INFO] [1516059892.140332175, 1515519542.620934904]: I0115 15:44:52.000000 12672 motion_filter.cc:42] Motion filter reduced the number of nodes to 15.6%.
[ INFO] [1516059893.972021512, 1515519547.017301169]: I0115 15:44:53.000000 12672 submaps.cc:295] Added submap 2
[ INFO] [1516059894.511402298, 1515519548.165072409]: I0115 15:44:54.000000 12672 offline_node.cc:183] Processed 32.4069 of 696.342 bag time seconds...
[ INFO] [1516059899.198204031, 1515519557.732330222]: I0115 15:44:59.000000 12672 submaps.cc:295] Added submap 3
F0115 15:44:59.205286 12672 local_trajectory_builder.cc:71] Check failed: !range_data.returns.empty() 
[FATAL] [1516059899.205516676, 1515519557.762312447]: F0115 15:44:59.000000 12672 local_trajectory_builder.cc:71] Check failed: !range_data.returns.empty() 
*** Check failure stack trace: ***
    @     0x7f6311ad35cd  google::LogMessage::Fail()
    @     0x7f6311ad5433  google::LogMessage::SendToLog()
    @     0x7f6311ad315b  google::LogMessage::Flush()
    @     0x7f6311ad5e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x6438cb  cartographer::mapping_3d::LocalTrajectoryBuilder::AddRangeData()
    @           0x5df408  cartographer::mapping::GlobalTrajectoryBuilder<>::AddSensorData()
    @           0x6d4595  cartographer::sensor::Dispatchable<>::AddToTrajectoryBuilder()
[... 54 lines cut ...]
Count: 69416  Min: 0.000024  Max: 0.421201  Mean: 0.010029
[0.000024, 0.042141)	####################	Count: 69402 (99.979836%)	Total: 69402 (99.979836%)
[0.042141, 0.084259)	                    	Count: 9 (0.012965%)	Total: 69411 (99.992798%)
[0.084259, 0.126377)	                    	Count: 0 (0.000000%)	Total: 69411 (99.992798%)
[0.126377, 0.168495)	                    	Count: 1 (0.001441%)	Total: 69412 (99.994240%)
[0.168495, 0.210612)	                    	Count: 0 (0.000000%)	Total: 69412 (99.994240%)
[0.210612, 0.252730)	                    	Count: 0 (0.000000%)	Total: 69412 (99.994240%)
[0.252730, 0.294848)	                    	Count: 0 (0.000000%)	Total: 69412 (99.994240%)
[0.294848, 0.336966)	                    	Count: 1 (0.001441%)	Total: 69413 (99.995682%)
[0.336966, 0.379083)	                    	Count: 1 (0.001441%)	Total: 69414 (99.997116%)
[0.379083, 0.421201]	                    	Count: 2 (0.002881%)	Total: 69416 (100.000000%)
E0115 17:03:07.598158 15326 rosbag_validate_main.cc:316] Point data (frame_id: "vertical_vlp16_link") has a large gap, largest is 0.925124 s, recommended is [0.0005, 0.05] s with no jitter.
I0115 17:03:07.609241 15326 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/vert_lidar_points" (frame_id: "vertical_vlp16_link"):
Count: 1030630  Min: 0.000001  Max: 0.925124  Mean: 0.000674
[0.000001, 0.092513)	####################	Count: 1030598 (99.996895%)	Total: 1030598 (99.996895%)
[... cut ...]
[0.370050, 0.462563)	                    	Count: 2 (0.000194%)	Total: 1030626 (99.999611%)
[0.462563, 0.555075)	                    	Count: 2 (0.000194%)	Total: 1030628 (99.999809%)
[0.555075, 0.647587)	                    	Count: 0 (0.000000%)	Total: 1030628 (99.999809%)
[0.647587, 0.740099)	                    	Count: 1 (0.000097%)	Total: 1030629 (99.999901%)
[0.740099, 0.832612)	                    	Count: 0 (0.000000%)	Total: 1030629 (99.999901%)
[0.832612, 0.925124]	                    	Count: 1 (0.000097%)	Total: 1030630 (100.000000%)
```

gaschler, 2018-01-23:
> W0115 17:02:19.435412 15326 rosbag_validate_main.cc:103] frame_id imu_link time 1515519546587567962: IMU linear acceleration is 2.23423 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0.0473449 002.21687 -0.273885

Please configure your IMU that it includes gravity. It won't work otherwise.
Actually, we only look at gravity direction and drop the rest of the acceleration vector.

> [FATAL] [1516059899.205516676, 1515519557.762312447]: F0115 15:44:59.000000 12672 local_trajectory_builder.cc:71] Check failed: !range_data.returns.empty() 

Yeah, this is a rather strict check.
You could replace it by a `return nullptr` in your fork or make sure your range data message are non-empty.

cschuet, 2018-01-24:
@ryanmeasel You could also try to leave the code untouched and simply filter out the empty point cloud ROS messages using a carefully crafted invocation of [rosbag filter(http://wiki.ros.org/rosbag/Commandline#filter) I guess :)

closed 2018-01-24 by cschuet