cartographer-project/cartographer #861: Special considerations for multiple loop closure in the same location

guilhermelawless, 2018-01-29:
Our use case is mapping a small apartment with a ground robot.

When using common tools (in ROS) such as gmapping, we commonly go through the same place multiple times; this helps reduce drift and overall error, and also give us insight while controlling the robot.

Disabling global SLAM I notice that I actually don't have a lot of drift (around 3% of the map size) after a long trajectory. However, when I enable global SLAM, every pass through the same location after the second adds additional error after loop closure.

Please see below. The upper area was where we only did one pass. The bottom right area is the most difficult area to map, where we made multiple passes. See for instance the right-most area, where every pass "added" a new map with drift. This was all done offline to make sure it is not due to the computer not keeping up.

[image]

I would like some general guidelines towards solving this, but if proven difficult, I can share more files.

Thanks!

guilhermelawless, 2018-01-29:
I realized it would become difficult to solve the issue without providing the bag. Please see below, and thank you so much for your time.

I have tried both with use_odometry true/false. The result does not improve.

Bag and config file: https://drive.google.com/open?id=1Jj13ybo3ngunO14AwcPXvsnpZbMAsglf

It needs a remap from /scan to /scan_front , apologies for the trouble.

gaschler, 2018-01-30:
First, please make sure that global SLAM really finished. It works in background threads.
In rviz, add topic "constraint_list" and visualize "inter" constraints. You should see *inter* constraints between almost all submaps nearby.

Set `-keep_running=true` for the node. Otherwise, it will not update rviz even though it did solve global slam.

guilhermelawless, 2018-02-01:
@gaschler  thank you, the result is much better after setting that option.

Is this a normal number of inter constraints? I thought it would be sparser after optimization.

[image]

Thank you!

gaschler, 2018-02-14:
Optimization never removes constraints that have been found.
You can search for fewer constraints by tuning the map builder options, which makes global SLAM faster (see the tuning guide).
Seeing the picture, I would try limit the laser ranges slightly shorter, perhaps this improves the quality.

(Closing as answered.)

closed 2018-02-14 by gaschler