cartographer-project/cartographer #882: Search window for finding loop closures should be gravity-aligned

ojura, 2018-02-02:
I had a dataset where I had some unwanted loop closures between similar floors. I tried to address this by reducing `MAP_BUILDER.pose_graph.constraint_builder.fast_correlative_scan_matcher_3d.linear_z_search_window`. To my puzzlement, this had no effect. 

However, when I changed the used tracking frame from the manufacturer's (z points backwards) to a standard ROS `base_link` frame with z upwards, the bad loop closures went away. 

I conclude that the setting I used for the z search window was applied in the z direction of the tracking frame, i.e. in the forward/backward direction. In the "up" direction, it seems that the larger `xy` search window was used, yielding bad loop closures.

ojura, 2018-02-03:
I have identified another place where an y-up tracking frame causes unexpected behaviour. 

It's the special cost function for the orientation of the first submap in 3D, which should result in constant yaw, but give the optimizer the freedom to change the pitch and roll of the first submap:
https://github.com/googlecartographer/cartographer/blob/df518b80c8c2cf8cb2e1ccd25b6309373e901044/cartographer/mapping_3d/pose_graph/optimization_problem.cc#L208

I have repeated a procedure where I add a constant, run optimization, remove the same constant, and re-run optimization, expecting the optimizer to produce the same solution as before the constant was added. Instead, I observed yaw drift of the complete trajectory as I repeated the procedure. My hypothesis is that   an y-up, z-backwards tracking frame again causes this, by making `ConstantYawQuaternionPlus` lock the angle around the wrong axis. Since the optimizer reuses the previous solution, and _real_ yaw is not locked, the computed solution trajectory's yaw is free to float and drift around.

ojura, 2018-02-05:
@wohe there's a good chance you wrote the code I mentioned above, would love to hear your opinion.

wohe, 2018-08-24:
@ojura Do you think this might be resolved by #1398? If so, could you verify?

ojura, 2018-08-27:
@wohe Not sure. I switched to using an intermediate Z-up frame for the tracking frame to mitigate this. I'll re-test sometime and let you know.

ojura, 2018-09-19:
@wohe I am certain that https://github.com/googlecartographer/cartographer_ros/issues/990 is an instance of this issue. You can see in the video that the submaps are oriented X-up.

ojura, 2018-09-19:
And pulling to the newest version resolved googlecartographer/cartographer_ros#990. Seems like we have a confirmation that this is now fixed. Sorry for the noise, @wohe.

closed 2018-09-19 by ojura

WEICHUAN1107, 2020-03-29:
@ojura The Z-axis of IMU need to be trasformed to the direction that parallel to the gravity and up?