cartographer-project/cartographer #910: Check fails for multiple laser scanners with per-point time

gaschler, 2018-02-16:


gaschler, 2018-02-16:
Here is a work around (not a real fix): https://github.com/googlecartographer/cartographer/pull/911

MichaelGrupp, 2018-02-17:
That's funny - I had to deal with the **exact same issue** yesterday - here' s my version of a quick fix: https://github.com/magazino/cartographer/commit/9f14301227ce525d32d49c6bfe26e447481127fc

gaschler, 2018-02-28:
Update: After the related fix https://github.com/googlecartographer/cartographer/pull/936, the check no longer fails.
But, time-overlapping points will only be unwarped with the latest extrapolator time (not the exact per-point time), which is not optimal, and you see a warning for each overlapping PointCloud.

gaschler, 2018-03-19:
Fixed by #996, closing.

closed 2018-03-19 by gaschler