cartographer-project/cartographer #934: Cartographer cmake error: ‘CellLimits’ in namespace ‘cartographer::mapping::proto’ does not name a type

gangyahaidao, 2018-02-26:
Hello everyone!
Install Cartographer have cmake error.  on ubuntu 16.04.9 , gcc 5.4.0, protoc 3.5.1

cmake ..
make -j8
then has the errors:

```
[ 14%] Building CXX object CMakeFiles/cartographer.dir/cartographer/mapping_2d/ray_casting.cc.o
In file included from /home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h:28:0,
                 from /home/rosrobot/git/cartographer/cartographer/mapping_2d/probability_grid.h:23,
                 from /home/rosrobot/git/cartographer/cartographer/mapping_2d/ray_casting.h:23,
                 from /home/rosrobot/git/cartographer/cartographer/mapping_2d/ray_casting.cc:17:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/xy_index.h:39:36: error: ‘CellLimits’ in namespace ‘cartographer::mapping::proto’ does not name a type
   explicit CellLimits(const proto::CellLimits& cell_limits)
                                    ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/xy_index.h: In constructor ‘cartographer::mapping::CellLimits::CellLimits(const int&)’:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/xy_index.h:40:33: error: request for member ‘num_x_cells’ in ‘cell_limits’, which is of non-class type ‘const int’
       : num_x_cells(cell_limits.num_x_cells()),
                                 ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/xy_index.h:41:33: error: request for member ‘num_y_cells’ in ‘cell_limits’, which is of non-class type ‘const int’
         num_y_cells(cell_limits.num_y_cells()) {}
                                 ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/xy_index.h: At global scope:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/xy_index.h:47:15: error: ‘CellLimits’ in namespace ‘cartographer::mapping::proto’ does not name a type
 inline proto::CellLimits ToProto(const CellLimits& cell_limits) {
               ^
In file included from /home/rosrobot/git/cartographer/cartographer/mapping_2d/probability_grid.h:23:0,
                 from /home/rosrobot/git/cartographer/cartographer/mapping_2d/ray_casting.h:23,
                 from /home/rosrobot/git/cartographer/cartographer/mapping_2d/ray_casting.cc:17:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h:50:35: error: ‘MapLimits’ in namespace ‘cartographer::mapping::proto’ does not name a type
   explicit MapLimits(const proto::MapLimits& map_limits)
                                   ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h: In constructor ‘cartographer::mapping::MapLimits::MapLimits(const int&)’:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h:51:32: error: request for member ‘resolution’ in ‘map_limits’, which is of non-class type ‘const int’
       : resolution_(map_limits.resolution()),
                                ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h:52:44: error: request for member ‘max’ in ‘map_limits’, which is of non-class type ‘const int’
         max_(transform::ToEigen(map_limits.max())),
                                            ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h:53:33: error: request for member ‘cell_limits’ in ‘map_limits’, which is of non-class type ‘const int’
         cell_limits_(map_limits.cell_limits()) {}
                                 ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h: At global scope:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/map_limits.h:92:15: error: ‘MapLimits’ in namespace ‘cartographer::mapping::proto’ does not name a type
 inline proto::MapLimits ToProto(const MapLimits& map_limits) {
               ^
In file included from /home/rosrobot/git/cartographer/cartographer/mapping_2d/ray_casting.h:23:0,
                 from /home/rosrobot/git/cartographer/cartographer/mapping_2d/ray_casting.cc:17:
/home/rosrobot/git/cartographer/cartographer/mapping_2d/probability_grid.h:34:41: error: ‘ProbabilityGrid’ in namespace ‘cartographer::mapping::proto’ does not name a type
   explicit ProbabilityGrid(const proto::ProbabilityGrid& proto);
                                         ^
/home/rosrobot/git/cartographer/cartographer/mapping_2d/probability_grid.h:73:10: error: ‘ProbabilityGrid’ in namespace ‘cartographer::mapping::proto’ does not name a type
   proto::ProbabilityGrid ToProto() const;
          ^
CMakeFiles/cartographer.dir/build.make:413: recipe for target 'CMakeFiles/cartographer.dir/cartographer/mapping_2d/ray_casting.cc.o' failed
make[2]: *** [CMakeFiles/cartographer.dir/cartographer/mapping_2d/ray_casting.cc.o] Error 1
CMakeFiles/Makefile2:1432: recipe for target 'CMakeFiles/cartographer.dir/all' failed
make[1]: *** [CMakeFiles/cartographer.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

```

half a month ago, i can compile success, i think this is because the commit after that,
Any help is appreciated!

gaschler, 2018-02-26:
Have you tried a build from scratch (deleting the build folder first or at least clean it)?
When protobuf files change, a build on top of an existing build will usually not work.

gangyahaidao, 2018-02-28:
@gaschler Yes,you are right, i compile Cartographer from scratch successful, thanks a lot!
Here are my compile records:
Mainly i follow this url:
[https://google-cartographer.readthedocs.io/en/latest/](url)
but on ubuntu16.04 i have some change:
1.change "https://ceres-solver.googlesource.com/ceres-solver" to "https://github.com/ceres-solver/ceres-solver.git"
2.change the version of protobuf 3.4.1 to 3.5.1
3.git clone https://github.com/googlecartographer/cartographer.git
use ninja to compile it
OK.

closed 2018-02-28 by gangyahaidao