cartographer-project/cartographer #937: HybridGrid and VoxelFilter fail with large coordinates

gaschler, 2018-02-27:
When a coordinate is 10000 voxels away from the origin, VoxelFilter or HybridGrid fail the check
```
F0227 09:00:59.550034 97228 hybrid_grid.h:387] Check failed: new_bits <= 8 (9 vs. 8) 
```

Even before that, performance degrades severely.
When a coordinate is 5000 voxels away, inserting a single value takes more than 100 ms (in Release mode).

```
TEST(HybridGridTest, LargeIndexes) {
  HybridGridBase<uint8> hybrid_grid(1.f);
  const Eigen::Vector3f large(0, 5000, 0);
  const Eigen::Array3i index = hybrid_grid.GetCellIndex(large);
  auto* value = hybrid_grid.mutable_value(index);
  *value = 1;
}
```

[ RUN      ] HybridGridTest.LargeIndexes
[       OK ] HybridGridTest.LargeIndexes (114 ms)

xiangkui123, 2018-02-28:
What is your testing environment?

cschuet, 2018-02-28:
We have had reports from people using Cartographer on cars who ran into this problem. This bug is however independent of the use case just analyzing the limitations of our voxel filter .

gaschler, 2018-03-02:
VoxelFilter is fixed and verified. HybridGrid still has limits, but they are now documented.

closed 2018-03-02 by gaschler

vaibhavviswanathan, 2018-04-18:
Hi, I just had this same issue using the latest cartographer clone.

gaschler, 2018-04-18:
@vaibhavviswanathan, can you please detail your issue? Ideally, can you give reproducible instructions?

vaibhavviswanathan, 2018-04-18:
I was able to solve the issue by changing my odometry sample ratio from 1.0 to 0.1.

What does this parameter mean?

ojura, 2018-04-18:
That 90% of your odometry measurements are dropped.

bochen87, 2018-07-18:
Hi, I'm getting the same problem with the `voxel_filter_and_remove_moving_objects` io point processor on a larger map (approx 600m x 300m) and a small voxel size (0.02)
https://github.com/googlecartographer/cartographer/blob/0668411d6fe3099ac663f700cbda407be0aae313/cartographer/io/outlier_removing_points_processor.h#L81

Would it be possible to change the implementation of this filter to not use the hybrid grid? I think it's a really useful point processor to get clean 3D pointcloud visualizations of the 3D map, but a pity if it doesn't work with bigger maps :-(

robokamran, 2021-07-19:
I have also the same problem on large maps (like a 60m x 40m place), with 3D point cloud, 12GB bag file, `voxel_filter_and_remove_moving_objects` with voxel filter of size 0.1 and `write_xray_image` with voxel filter of size 0.01.
It is ok on smaller maps, the problem is with large map. Points2 rate is 10Hz and IMU data rate is 200Hz. It is also ok with just creating ply file and not the x-ray images so I assume that the problem is not with voxel filter of the moving-object-remover, but the x-ray image generator.